fraschetti/Octoslack #4: [feature request] send out messages to riot (matrix)

arthurlutz, 2017-05-29:
I know that the plugin contains slack in the name, but since it also can be used with mattermost, I going to try out my luck with this feature request : it would awesome for this plugin to be able to talk to riot https://riot.im/ which is a open source / self-hosted / privacy oriented alternative to slack. 

This would probably be done via the matrix python client https://github.com/matrix-org/matrix-python-sdk 

I'm trying out the matrix/riot/slack integration to get messages from octoprint -> slack -> riot (but this is a hack and right now doesn't work yet)

fraschetti, 2019-06-20:
Hi @arthurlutz 

It only took me 2 years to get around to this issue but as of last night and Octoslack 1.8.1, you finally have your Riot (Matrix) support.

closed 2019-06-20 by fraschetti

arthurlutz, 2019-06-21:
@fraschetti thanks a lot ! The list of channels looks awesome! Time to rename the plugin ? ;-)

thearduinoguy, 2019-07-22:
How do I get the access token from Riot please? The instructions presume you already have a bot set up.