fraschetti/Octoslack #5: Not posting

Vikinge, 2017-05-30:
When I reboot my octopi, I see octoslack post 

> octopi
> -Octoprint service started (check mark)
> 

But nothing else appears, no start, no failure, nothing.  I can run @octopi status and this returns thumbs up and a 1, but no more information.  
Any suggestions for troubleshooting.  I am using the API and have BOT defined for my slack.

mkotek, 2017-06-03:
first of all, status command depends on Progress event, that is not enabled in default configuration. Check the settings for octoslack plugin in the configuration and set the event to Enabled.

In order to check if plugin is responding, without changing anything, you can ask your bot help question and you should get the answer. Also, if the command is not understood, it will respond with thumbs down, so at least you know, something is responding :)

fraschetti, 2017-06-07:
Can you confirm the channel you've configured the plugin to use and/or the bot to use matches the channel where you're executing the @octopi commands?

Vikinge, 2017-06-08:
Updating to your recent release 1.1.1, had newer API configuration, that
solved my issues with IMGUR.  Everything seems to be working fine now.
Thank you for your help.

fraschetti, 2017-06-09:
Glad to hear it; please let me know if you run into any other issues.

closed 2017-06-09 by fraschetti

fraschetti, 2017-06-11:
FYI - I also recently found (and fixed) a bug where "@bot status" wouldn't actually report back the status if the "Progress" event hadn't been checked. I've fixed this for a future build but enabling the "Progress" event on the plugin.