fraschetti/Octoslack #11: Json Data

CamoJackson, 2017-09-27:
is there any way to add a flag to pause which will read json data
example
i want to add this to my pause report using integration with the Filament Sensor plugin by ovidiu
```
{filament} = octopi.local/plugin/filament/status?apikey=**
If filament present: {filament}
```

fraschetti, 2017-11-12:
Hi @CamoJackson ,

It's certainly possible to do something like that.
After a few minutes of thinking about this, I'd likely implement it as a generic "shell command --> variable" feature that requires whatever command you run to dump return the exact text you want to standard out (the console) for Octoslack to consume.
To fetch your filament value, you'd just need to write a shell script that leverages curl or whatever other mechanism you choose.

If I were to implement this, there would be a few limits I'd need to put in place:
1. How long I let the script run (we can't hang up OctoPrint waiting for a script to return - likely only a seconds at the most)
2. How much text we let the script return (likely only a few hundred characters)
3. As with another feature I have in mind, providing Octoslack a user configurable shell command input is a bit of a security risk, particularly for those folks who decide to put their OctoPrint instance out on the public web.

How does that sound to you?

fraschetti, 2018-11-09:
It's been (almost) a year without any activity on this issue. I'll close it out until the request comes up again.

closed 2018-11-09 by fraschetti