fraschetti/Octoslack #13: Images do not respect the OctoPrint webcam settings

jb1t, 2017-10-30:
I'm running version 1.0.1 of OctoSlack. I'm also running version Version 1.3.5 (master branch) of OctoPrint. I'm running on a RPi 2 model b. 

As you can see below I've set the webcam setting to flip both horizontally and vertically. My webcam works great and so does my timelapse videos. However, the images posted to imgur and then being referenced at Mattermost are not respecting these settings. 

Any suggestions? 

Thanks 👍 

[image]
[image]

fraschetti, 2017-11-09:
I've just submitted a few debug log items to help diagnose what might be going on here.
For now they're in the sitting in the master branch until I publish a build.

Once you have a build installed with these changes, you can update your OctoPrint logging.yaml (located in ~/.octoprint when using OctoPi) with the following lines and then restart OctoPrint (sudo octoprint restart):

  octoprint.plugins.Octoslack:
    level: DEBUG

A healthy output executing all image manipulations would look like this:

Local camera settings - Snapshot URL:http://127.0.0.1:8080/?action=snapshot, FlipH: True, FlipV: True, Rotate90: True
Snapshot URLs: [('http://127.0.0.1:8080/?action=snapshot', True, True, True)]
Downloaded snapshot from URL: http://127.0.0.1:8080/?action=snapshot (21.7KB) in 0.06 seconds to /tmp/tmpqoUPVE
Transpose operations for URL: http://127.0.0.1:8080/?action=snapshot - FlipH: True, FlipV: True, Rotate90: True
Opening file to transpose image for URL: http://127.0.0.1:8080/?action=snapshot
Flipping image horizontally for URL: http://127.0.0.1:8080/?action=snapshot
Horizontally flip complete for URL: http://127.0.0.1:8080/?action=snapshot
Flipping image vertically for URL: http://127.0.0.1:8080/?action=snapshot
Vertical flip complete for URL: http://127.0.0.1:8080/?action=snapshot
Rotating image 90 degrees for URL: http://127.0.0.1:8080/?action=snapshot
90 degree rotate complete for URL: http://127.0.0.1:8080/?action=snapshot
Saving transposed image for URL: http://127.0.0.1:8080/?action=snapshot to /tmp/tmpqoUPVE
Downloaded all 1 snapshots in 0.37 seconds

fraschetti, 2017-11-11:
1.4.0 has been officially released and includes the previously mentioned debugging logic,

fraschetti, 2017-11-12:
Hi @jb1t ,

I absolutely overlooked the fact that you're running Octoslack 1.0.1.
Octoslack's support for respecting OctoPrint's camera orientation settings was added in 1.1.0

Unfortunately the early versions of Octoslack had a typo in one of the properties files which prevented the OctoPrint from detecting/installing new releases as they come out.
The solution here would be to uninstall Octoslack and reinstall it which will get you the latest and greatest (currently 1.4.0).

Good luck and please let me know if you this doesn't solve your problem.

closed 2017-11-12 by fraschetti