fraschetti/Octoslack #14: Send message when heating is finished and printing starts

stepjanssen, 2017-11-13:
I'd like to be able to receive a message when printing is started after pre-heating is done. I now get a message when I click print, but then I keep checking the camera to see when the actual first layer starts printing.

- Stephan

fraschetti, 2017-11-13:
Hi @stepjanssen,

The stock set of events in Octoslack come from hooks provided by OctoPrint.
As far as OctoPrint is concerned, once your g-code starts running, that's technically the 'start' of the print.
The easiest way to accomplish what you described is to inject a bit of custom g-code into your slicer's start g-code template (assuming your slicer support's this if course). From there you can use the G-code event feature introduced in Octoslack 1.4.0 to trigger your event.

As an example, here's my Slic3r 'Start G-code' template; the bold line is where I'd personally inject my 'print has actually started' g-code (basically at the end).

`M115 U3.0.12 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 
**M117 Print in progress**

From there you can simply create a G-code event in Octoslack that looks for **M117 Print in progress**

stepjanssen, 2017-11-15:
@fraschetti, many thanks for the clear explanation! Will be trying this on my next print.

fraschetti, 2017-11-16:
Sounds good. Let me know if you encounter any issues.

closed 2017-11-16 by fraschetti