fraschetti/Octoslack #15: Not able to see location to enter api token in settings

brbjr1, 2017-12-15:
Octoipi version 1.3.6

In the chrome dev console when loading home page is see error: 
packed_core.js?2691f4a1:13555 Error calling onAfterBinding on view model OctoslackViewModel : ReferenceError: Octoslack is not defined
    at HTMLInputElement.onclick (http://192.168.192.133/#temp:5415:202)
    at Object.trigger (http://192.168.192.133/static/webassets/packed_libs.js?eebed41b:7814:27)
    at HTMLInputElement.<anonymous> (http://192.168.192.133/static/webassets/packed_libs.js?eebed41b:7876:17)
    at Function.each (http://192.168.192.133/static/webassets/packed_libs.js?eebed41b:366:19)
    at jQuery.fn.init.each (http://192.168.192.133/static/webassets/packed_libs.js?eebed41b:138:17)
    at jQuery.fn.init.trigger (http://192.168.192.133/static/webassets/packed_libs.js?eebed41b:7875:15)
    at Object.setInitialInputStates (http://192.168.192.133/static/webassets/packed_plugins.js?a14fe561:66:34)
    at Object.afterBindingInit (http://192.168.192.133/static/webassets/packed_plugins.js?a14fe561:49:22)
    at OctoslackViewModel.self.onAfterBinding (http://192.168.192.133/static/webassets/packed_plugins.js?a14fe561:20:31)
    at callViewModelIf (http://192.168.192.133/static/webassets/packed_core.js?2691f4a1:13545:34)

MStylesMS, 2017-12-15:
I also have this issue.  There is no place to enter the Slack API Token, Slack WebHook URL, or the Imgur account login.

fraschetti, 2017-12-16:
Thanks for reporting the issue; I'll take a look asap.

fraschetti, 2017-12-16:
This was a quick fix. I've published 1.4.1 which you can wait for OctoPrint to prompt you to upgrade or you can force a check now via the plugin manager.

NOTE: anyone on Octoslack 1.2.0 or earlier are impacted by a plugin upgrade bug and should manually uninstall and then reinstall to get on the latest and greatest.

closed 2017-12-16 by fraschetti