fraschetti/Octoslack #16: Unable to set S3 Snapshot Uploads

slintak, 2017-12-18:
This plugin is really great! I am currently using it on two different printers and events itself are very usefull. No I want to set snapshots...

I created new S3 bucket to my Amazon account and created new user with full access permissions to S3 resource. Any event send to Slack now contains snapshot error: `S3 error: 403 Client Error: Forbidden for url: https://*****/Snapshot_b65ea01ee40911e785d8b827eb46c4d3.png`.

What is the cause of this error? Can you, please, add more informations about S3 settings? How to set permissions/policy on bucket?

Lets say I create new IAM user with name `octoprint` and its AWS Access Key is `ABCD` and AWS Secret key is `foo/bar`. Now I create new bucket with name `octoprint-snapshots`. I save these keys to Octoslack settings. What now? What permissions should I add to bucket and/or IAM user? How can I test everything works without actually waiting for some event (i.e. how to trigger some event from CLI).

fraschetti, 2017-12-19:
I'll take a look. Can you share your OctoPrint + Octoslack versions so I know what we're working with?

fraschetti, 2017-12-19:
Hi @slintak 
From the Octoslack perspective and ultimately the AWS S3 AP perspective, the access key + secret key + bucket are all that can be configured.
From there you'll need to ensure your permissions are set correctly on the bucket itself for the user you created to access it.
You'll need to play around in your S3 console but I suspect you'll want to visit the Permissions tab when editing your bucket to ensure the user whose keys you've provided to Octoslack actually has the proper read/write permissions.

Hope this helps.

closed 2017-12-19 by fraschetti

edsvcs, 2018-04-02:
@slintak Did you get this to work? I just setup  Octoslack and am running into the same error message. I appreciate any pointers you can share.

edsvcs, 2018-04-02:
For testing I granted some pretty broad IAM permissions and now OctoPrint / Ocoslack is creating image files in my S3 bucket and I no longer see the error messages in Slack but I also don't get pictures or links in the Slack messages... any ideas? I have deslected and reselected the check boxes in Octoslack indicating that snapshots should be included.

slintak, 2018-04-02:
Hello @edsvcs. I ended up with the same problem as you. Photos are stored in S3, but Slack won't display them. I am using Slack webhooks.

edsvcs, 2018-04-02:
Thanks @slintak . I opened a new issue. Hopefully between the two of us we'll come up with a solution.