fraschetti/Octoslack #17: Update 1.4.0 -> 1.4.1 not possible

MarphXL, 2017-12-19:
Have OctoPrint v1.3.6 on a raspberry pi jessie lite and Octoslack v1.4.0.
Getting the Update-Message for OctoSlack v1.4.1
After installing via Webinterface, OctoPi installed succesfully, reboot and I have only v1.4.0 and get the updatemessage to 1.4.1.
Try it for several times.
What should I do?

fraschetti, 2017-12-19:
Hi @MarphXL 
The upgrade likely succeeded but in my rush to get the 1.3.6 out the door, I failed to bump the internal Octoslack version in the code to 1.4.1. The end result is OctoPrint thinking Octoslack is still on 1.4.0 but that git has a 1.4.1 available.
I'll be pushing another release shortly which won't have any new functionality but should take care of this upgrade issue.

MarphXL, 2017-12-19:
great Chris! And good Job. It's a fantastic plugin

fraschetti, 2017-12-20:
I've just released 1.4.2 which should fix this upgrade issue.
Thanks!

closed 2017-12-20 by fraschetti