fraschetti/Octoslack #18: Printing crippled by incorrect IMGUR settings.

MStylesMS, 2017-12-29:
First of all, thanks for a great plug-in that I needed!  I hope this feedback helps.

So first, the instructions for IMGUR image hosting don't follow what is actually needed.  In the instructions you provide a link to authorize the application but it generates the following error:

"data":{"error":"HTTP Access is disabled. Requests must use SSL (HTTPS)."},"success":false,"status":400}

The issue appears to be that your link does not include "https://" at the front of it so it is being rejected by IMGUR.  

The second issue is that in following your instructions I am never able to find the IMGUR refresh token.  I am able to create the client Id and secret, but no refresh token.  

The result is that printing is slowed down by Slack messages like the following the appear to be sent after EVERY g-code command:

  *File* Layer_Test-S-20-55-125.gcode (35.9KB via OctoPrint)
  *Snapshot error(s):*
  *-* Imgur error: No album was found with the ID: 3dprinter
  Printer: Operational, Bed: 62.7°C/0.0°C, Nozzle: 157.0°C/0.0°C, RasPi: 52.6°C

Obviously such errors should not cripple Octoprint to the point of being unusable.  

Also, please note that the process for getting the Slack API token is also not exactly as you described.  Likely they changed the navigation on their site, but I had to do a good bit of searching to find the proper page.

MStylesMS, 2017-12-29:
Also, I just noticed that there was a setting at the bottom that looked blank but was actually producing a notice upon every OK response, which also killed the printing.  While that was a user error on my part, it might be nice to filter and produce a warning for such situations.  I don't even know when or how that was set.

fraschetti, 2018-01-23:
Hi @MStylesMS 

I've submitted an fix for the Imgur link (HTTP --> HTTPS) for a future release. It does indeed look like their site stopped supporting HTTP.

Can you share a few more details on the setting that was generating so many Slack messages?

MStylesMS, 2018-01-23:
As I remember, there wasn’t a particular setting that caused it.  What the problem appeared to be was that it kept trying to post to IMGUR and for some reason was posting a message about the failed post to Slack.  Unfortunately I deleted that channel from Slack because it had several hundred (maybe over a thousand) messages that were taking up the capacity of 10,000 messages I had at the time.

Mark Stevens
Co-Founder & Vice-President
[cid:08AF14B4-CBFC-4883-8A6D-CCE15200A5C4]

fraschetti, 2018-01-25:
Hi @MStylesMS 

I was able to easily reproduce the G-code event issue by saving an entry with a blank 'G-code' field.
I've just published Octoslack 1.4.3 which fixes the Imgur addclient link (switched to HTTPS) but also will ignore these blank G-code events.

Thanks for submitting this issue.

closed 2018-01-25 by fraschetti

MStylesMS, 2018-01-25:
Thank you for fixing it!