fraschetti/Octoslack #19: Extra variables

scns, 2018-03-03:
Hi,

Is it possible to get an variable for the actual layer number and the actual time?

Kind regards

fraschetti, 2018-03-04:
Hi @scns ,

1. By "layer number" are you referring to the current Z-height or a slice derived layer number? Exposing the current Z height as a variable shouldn't be difficult - assuming that's what you're looking for.

2. For "actual time" - can you be a bit more specific? Are you looking for the elapsed time? time remaining? estimated total print time?
The library already provides:
{elapsed_time} - Time since start of print 
{remaining_time} - Estimated remaining print time

So perhaps you're looking for the estimated total print time?

scns, 2018-03-04:
With the layer number i mean layer 86 of 126.

See image in OctoPrint:

[image]

And the actual time I just mean he system time of the raspberry.

But the layer number should be very nice if that is possible!

fraschetti, 2018-11-09:
Hi @scns 

Just to let you know - I've not forgotten about this request.
I did look into this immediately after you submitted your previous response but never found the time to post my findings here.
Unfortunately OctoPrint's layer calculation logic is all client-side and isn't available to our plugins without some serious hacks that would be unreliable to say the least.

The alternative solution I believe others have adopted here is to have the slicer insert a custom gcode when each new layer has started. In the case of Slic3r, it would make use of [layer_num]

Still brainstorming. I'll provide another update if anything changes.

fraschetti, 2018-11-09:
Also, {eta} was added in 1.6.0 (just released) which may satisfy your "actual time" request.

scns, 2018-11-13:
Tnx for the update and making the {eta} available. I will look for another option to get the Layer_num available.

closed 2018-11-13 by scns