fraschetti/Octoslack #20: Setup question with S3

edsvcs, 2018-04-02:
Cool plugin. I'm looking forward to using it fully!

Initially I was have a problem where no images were being created in my S3 bucket and Slack would display an error message "- S3 error: 403 Client Error: Forbidden for url: https://xxxxxxxxx"  with each message from Octoslack. I changed my bucket permissions to match a public web site, meaning that anyone who has a file URL can display it, and I have confirmed that if I put one of those URL's into Slack it displays. I also expanded IAM permissions assigned to Octoslack to basically make it a superuser figuring I will lock it down once it works and I understand which permissions it actually needs. Now Octoslack places images in the bucket but the Slack messages no longer contain links to the images. I have checked and unchecked the options to attach snapshots to messages with no effect.  I'm stumped. What am I missing or what information do you need to troubleshoot?

slintak, 2018-04-03:
I had similar issue with S3 (#16 ). I was able to set up AWS permissions and now Octoslack uploads images as expected. Now I have the same problem as @edsvcs -- no image snapshot in Slack message. Octoslack is configured to use webhooks and images in S3 can be displayed by anyone with URL.

edsvcs, 2018-04-10:
Just a follow up on this. I redid my integration using a bot this time. I am using the API because the on-screen instructions indicate some functionality might not work with web hooks, but still no images.

fraschetti, 2018-11-09:
As mentioned in the pull request referenced above, there may be two variations that I need to ensure Octoslack supports. I've just finished getting 1.6.0 out the door and pendind any major issues, I'll start looking into the changes @slintak has submitted in hopes of getting in a fix that works for everyone.

steckdev, 2018-11-15:
Thanks so much for your effort on this. I was fighting with the issue of too many pictures on Imgur.

fraschetti, 2019-02-21:
I've just pushed 1.7.2 which includes a drop-down option to switch between the two Amazon S3 URL styles. It's rare that this should matter but if your bucket meets one of a limited number of critiera, you might need to use one or the other.
Feel free to reopen this issue if you continue to have problems.

closed 2019-02-21 by fraschetti