fraschetti/Octoslack #21: Slack configuration question

edsvcs, 2018-04-02:
This is the first time I have added any services to Slack so I apologize in advance for the newbie question. I produced a legacy API key and added that to Octoslack and it is posting messages and responding to messages, but the messages all appear to be from me, so I am never alerted to them. From what I read about Bots it did not make sense to create a Bot and give that API key to Octoslack (they parse every message in a channel, not just those directed to them, etc...).... What is the recommended best practice? Should I create a dummy user as a surrogate for Octoslack and create an API key under that account?

Thanks for your help, and for providing this plugin!

hanzov69, 2018-04-04:
@edsvcs I ran in to a similar issue. It's non-obvious, but here is a quick explanation of what you need to do:

Create the application in slack api
create a bot user
install the app
then go back to install app, and you should see a bot user oauth token, will begin with xoxb-

as a cheat, this link (might) work to take you where you need to be: https://api.slack.com/apps/AA102T3FX/install-on-team?

edsvcs, 2018-04-10:
Thank you @hanzov69. This worked in terms of getting Octoslack to post as a bot instead of as me. I still don't get any images included in the messages even though they are captured. Did you get that part to work?

hanzov69, 2018-04-10:
@edsvcs I did, I just went down the imgur path. What was odd was the the first time it attempted to post an image, it was just a black field, and nothing was uploaded to imgur. Restarting the octopi (for unrelated reasons) seemed to fix it right up though.

fraschetti, 2018-11-09:
Hi @edsvcs 

Are you still having issues? If so, let me know. If not, I'll close out this issue.

Thanks!

fraschetti, 2019-02-18:
This issue is rather old; feel free to repoen it if your issue surfaces again.

closed 2019-02-18 by fraschetti