fraschetti/Octoslack #25: suggestion: timed status when not printing

uwezi, 2018-09-03:
I would appreciate if there was an option to send regular status messages from the bot even when the printer is not printing - just to see that everything is ok...

fraschetti, 2018-10-13:
Thanks for the suggestion, I'll add it to the list.

fraschetti, 2018-10-31:
Hi @uwezi 
This logic is in and it'll be included in the next update.
Keep an eye out for a new "Printer status heartbeat" event which can be configured to send a message every N minutes

fraschetti, 2018-11-09:
This functionality is now available in 1.6.0

Curently the heartbeat is active even when printing. Let me know if that's an annoyance I can consider halting the heartbeats while a print is in progress.

closed 2018-11-09 by fraschetti