fraschetti/Octoslack #26: [issue?] bot away

uwezi, 2018-09-03:
I managed to configure the plugin for three 3D-printers running octoprint. I even got imgur running for my own printer but not yet for the two printers at work.

However, I had some issues(?):

- two of the bots suddenly disappeared from slack and I had to re-configure the bots
- all three bots stopped answering to status queries - is this only active while octoprint is printing?
- all bots are "away" in slack

fraschetti, 2018-11-08:
Hi @uwezi 

> * two of the bots suddenly disappeared from slack and I had to re-configure the bots

[Fraschetti] I'm not sure what happened here. From my side of things, the bot itself doesn't have permissions to remove the bot nor is there any logic in the pluging that would even try.

> * all three bots stopped answering to status queries - is this only active while octoprint is printing?

[Fraschetti] Do you mind enabling DEBUG logging for the plugin and then reporting back with your octoprint.log once you've seen the issue happen again? I've also added a bit more logging to this connection logic for the upcoming release. You might want to wait until that new version comes out so we both have the best chance of diagnosing your issue.
* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml and then restart OctoPrint
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

> * all bots are "away" in slack

[Fraschetti] Do you always see this when the bots no longer respond or do they also show as inactive even when the bots are respoding to queries?

It might also be helpful if you can report your OctoPrint version, Octoslack version, operating system, etc.

Thanks!

fraschetti, 2019-01-09:
Hi @uwezi 

Any chance you found time to look at my responses/questions?

Minhyme, 2019-01-12:
Hey @fraschetti, my bot seems to go offline after a while of idling. It will still send status updates every 25% per my settings but it will not respond to requests. It will only come back online after restarting my octopi. I will enable debugging logging and get back to you.

fraschetti, 2019-01-17:
Hi @Minhyme 
If you can provide even a single Octoprint log file and prvoide me with the context of what timestamps you believe the bot was online/offline, I'd appreciate it.
I'd first like to see what the log file might be telling us and if it's ultimately not very useful, I'll look into adding additional logging to help diagnose the issue.

Minhyme, 2019-01-18:
Hey @fraschetti,

I will need to setup an experiment tomorrow to print and then check slack periodically to determine time stamps. What's the best way to get you a log file? Will respond tomorrow once I have the timing on when the bot goes offline.

fraschetti, 2019-01-18:
Hi @Minhyme,

You should be able to attach the log file here. If you run into issues, feel free to just send it to me directly - fraschetti.dev@gmail.com

Minhyme, 2019-01-19:
Hey @fraschetti,

Please see the log file attached. I started with a system restart and then used the printer as I normally would. First checked the bot at 13:00 and was online. Checked the bot every half an hour until it went offline. Completed one print and started another in that time. Bot went offline some time between 17:00 and 17:28. Sent an "@printer status" at 17:38 through slack with no response. Bot successfully posted 75% status update and screenshot at 17:44.

[octoprint.log](https://github.com/fraschetti/Octoslack/files/2774955/octoprint.log)

I took a glance at the log file and didn't see too much useful information. Please let me know if I can help further.

fraschetti, 2019-01-19:
Thanks @Minhyme 
As you suspected, that log file didn't have anything that suggests what might be going on.
I've been digging deeper into library I'm using for Slack connectivity and it looks like there may be some deeper level functionality to detect stale connections, reconnect automatically, etc. - I'm working on that logic now. I'll likely let those changes bake here locally for a small while before I push out a new build.

fraschetti, 2019-01-26:
@Minhyme 
I just published 1.7.0 which includes includes the reconnect logic I previously mentioned as well as a hearbeat/ping that will attempt to keep the connection alive (Slack actually states this should happen when a client is idle). I'll leave this issue open for now - please report back once you've upgraded and have had a chance to see if the issue has been resolved.

Minhyme, 2019-01-27:
Hey @fraschetti,

I've upgraded and tested the new build. I haven't experienced any disconnecting. During my testing last time, the bot was online between 4 and 4.5 hours. So far my uptime is 20 hours and the bot is still going strong! 

Also I am very happy with the new feature of being able to define custom messages off of gcode snippets. Monitoring for crashes with my Prusa i3 MK3 remotely is now easy!

fraschetti, 2019-02-09:
Hi Folks,

I have a few Bot/RTM API changes baking that should make the connections even more resilient (particularly for our windows users).
The core issue here seems to have been addresses so I'm going to close this issue. Please feel free to report any future issues you might encounter.

closed 2019-02-09 by fraschetti