fraschetti/Octoslack #28: [Feature Request] Upload rendered timelapse

JoKneeMo, 2018-09-20:
When a timelapse render is completed, the only configuration option is to post the screenshot to the slack channel. Would it be possible to have it post the rendered video?

For the record, I'm using Amazon S3 for uploads and Rocket.Chat as the Slack API endpoint.

I imagine it doesn't already do it because imgur is the default uploader.

fraschetti, 2018-10-13:
Hi @JoKneeMo,

Entirely possible. I played a bit with attempting to render the snapshots to an animated gif but apparently Python (the underlying language for OctoPrint and Octoslack) is a bit lacking in this department.
More specifically, it can be done but the libraries required to make it happen are not precompiled for the RaspberryPi architecture. As such, installing a plugin that uses these libraries would include a multi-hour compile time which is a tad unreasonable.

Time permitting, I'll browse around to see if Python has a video library that may be more readily available.

JoKneeMo, 2018-10-14:
I looked at the code further and see that the snapshots are taken directly by the plugin.
While making a GIF of the collected snapshots sounds interesting, I wouldn't find it useful for my use cases.

My use case is to just upload the timelapse already rendered by octopi to Amazon S3.
It looks like the `MovieDone` event includes the full path to the file as `movie`.

I'm not that great at coding myself so I failed miserably, but what I was trying to do was
* Add a checkbox to the MovieDone settings to "Upload Video" in `Octoslack.js`
* Update `__init__.py` with `def upload_video(self)` with None and S3 as the options.
  * I got it to grab the video and upload it to the s3 bucket by setting its default to true
    * This was because I was unable to get the interface to show a configurable option due to the way they are pulled in to the page.
  * I was also unable to figure out how to get it to append the URL to the slack post for this event.

fraschetti, 2018-10-31:
Hi @JoKneeMo 
I don't personally use the timelapses because they don't (yet) support multiple cameras. Because of that, I wasn't even thinking of the OctoPrint timelapse feature when I originally sent my response.
I'll play around with this a bit - The Octoslack side of things isn't terribly complex but I want to see how a long S3 upload might behave within the OctoPrint ecosystem.

fraschetti, 2018-11-09:
This functionality is now available in 1.6.0

closed 2018-11-09 by fraschetti