fraschetti/Octoslack #30: Feature Request - Post notificaiton with Firmware triggers

jalanjarosz, 2018-10-12:
I have a Marlin Firmware based runout sensor.  Marlin triggers a M600 which pauses the print, moves the head and sets a timer for head cool-down until resumed by the end-user.

Marlin sends the following to the terminal screen when an event occurs: **echo:enqueueing "M600"**

Is it possible for OctoSlack to monitor for "received" events and send notification, like the  GCode setup?

fraschetti, 2018-10-13:
Hi @jalanjarosz,

Possibly. I'm overdue to upgrade my Prusa MK2 (upgrade kit(s) already in hand). I can likely give this a closer look once I have my local printer upgraded.

fraschetti, 2019-01-17:
Hi @jalanjarosz 

I've completed development of this enhancement and it'll go out the next time I push a new release (TBD).
Specific to catching the output for the filament runout sensor (M600), below you can see the relevant statements sent from my Prusa MK2.5 firmware and received by OctoPrint. My personal preference woud likely be a G-code received event that looks for "**fsensor_update - M600**". 
> fsensor_update - M600
> echo:Enqueing to the front: "FSENSOR_RECOVER"
> echo:Enqueing to the front: "M600"
> echo:busy: paused for user

As an added bonus, I decided to evolve the G-code event logic beyond the hard-coded "starts with" functionality to include:

- Starts with
- Ends with
- Contains
- Regular expression (evaluate the text as a regular expression against the sent/received G-code)

fraschetti, 2019-01-26:
1.7.0 was just released with this enhancement. Have fun!

closed 2019-01-26 by fraschetti

jalanjarosz, 2019-02-14:
fraschetti -
Thank you for the enhancement!  I was finally able to trigger a runout trigger and it worked like a charm!  Great job!  Keep up the awesome work on the plug-in.