fraschetti/Octoslack #37: WebDav Integration

calonmerc, 2018-11-15:
Love everything about the plugin, currently using it with Imgur. Would love if there was an implementation for the WebDav protocol for OpenSource storage software like Owncloud/Nextcloud.

fraschetti, 2018-11-16:
Hi @calonmerc 
Thanks for the suggestion.
I've never personally used the software you mentioned.
It may be straightforward enough to upload snapshots/timelapses via the webdav protocol but surfacing them via Slack may be a bit more tricky.
Can you share some additional context as to what you're hoping to happen once the file(s) have been uploaded? 
Is there a mechanism to expose them via HTTP/S?
I've yet to give it a try (should be easy enough) but I've yet to test how Slack handles locally (I presume on your home network vs. the public internet?) hosted image URL.  I suspect the Slack servers on their backend need access to the file for it to render in Slack itself.

calonmerc, 2018-11-16:
My Nextcloud instance is publicly accessible, as I imagine most use cases are. It's similar to a NAS, but meant to be a replacement for Dropbox, Google Drive, or similar. 

You can set public permissions on the file to be accessible from the web directly, but I don't know how much of that is surfaced directly in the WebDAV protocol. It may be something I need to investigate simply as an alternative upload location for timelapse images/videos in a separate plugin. Just wondered if it might work better with this implementation, as I would like them to also show up in Slack...

calonmerc, 2018-11-26:
I've spent some time looking into this, and found that it should be possible using the OCS API to enable a publicly shared folder by link. Unsure of the development requirements, or how else I can help integrate (I can give you some space on my server for testing, if you'd like).

https://docs.nextcloud.com/server/12/developer_manual/core/ocs-share-api.html

fraschetti, 2019-01-09:
Hi @calonmerc 

I spent a few minutes tonight looking at what could be done here.

- There's an existing Python (the language used for OctoPrint + Octoslack) library for WebDav (https://github.com/ezhov-evgeny/webdav-client-python-3) which seems like it should get the job done.

- Constructing the URL for the uploaded asset is fairly straighforward - _webdav server:port + path + asset_  (with maybe an override to allow uploads via a private host vs a public host for URL generation)

- My Synology NAS has a WebDav server I used for some initial testing but this is on my private network and inaccessible by the Slack servers (to have Slack render content, it needs to be able to fetch it from their servers). Do you mind uploading an image to your NextCloud instance and pasting that URL into Slack to see if Slack can fetch the asset (and render the thumbnail as it typically would)?

calonmerc, 2019-04-18:
Sorry for taking quite a while to get back to this!

Sadly, I am not seeing images being directly view-able without logging in or setting up link sharing. If I set the directory they're uploading to as shared via a link with no password, then any of the images show up in Slack just fine!

calonmerc, 2019-04-18:
Here is a directory that is link shared with no restrictions:
[Directory](https://cloud.hfd.xyz/s/WkNG2na836KNwxW)
[Sample Timelapse](https://cloud.hfd.xyz/s/WkNG2na836KNwxW/download?path=%2F&files=xyzCalibration_blank_20181206154858.mpg)
[Sample Image](https://cloud.hfd.xyz/s/WkNG2na836KNwxW/download?path=%2F&files=snapshot%20(1).jpg)