fraschetti/Octoslack #38: [Feature Request]

andre726, 2018-11-21:
Hello it's possible to do this plugin compatible with rocket chat?

fraschetti, 2019-01-09:
Hi @andre726,

I took a look at Rocket Chat and they do have a similar webhook feature to Slack + Mattermost.
Unfortunately Rocket Chat's expected incoming webhook postMessage format differs from Slack (and Mattermost who mimics the Slack format) and therefore a bit of extra work is required to make this work for Rocket Chat. I'll keep this request open and will likely tackle it as some point. If you have any specific requests or have an urgency for this support, please let me know.

fraschetti, 2019-06-20:
Hi @andre726 

As of 1.8.0, Octoslack now has support for Rocket.Chat. Please let me know if you encounter any issues.

closed 2019-06-20 by fraschetti