fraschetti/Octoslack #41: 'ETA' is always 'now'

tedder, 2018-12-25:
Let me know what info you need to reproduce, but here are some examples.

sent at 12:02:
> [12:02 AM]
> :heavy_minus_sign: Print progress 45% :horse_racing:
> *File* Stand_wMotorMount_0.2mm_PLA_MK3MMU2.gcode (18.6MB via OctoPrint)
> *Elapsed* 05:11:39
> *Remaining* 05:30:25
> *ETA* 00:02 today

sent at 12:33:
> [12:02 AM]
> :heavy_minus_sign: Print progress 50% :horse_racing:
> *File* Stand_wMotorMount_0.2mm_PLA_MK3MMU2.gcode (18.6MB via OctoPrint)
> *Elapsed* 05:42:54
> *Remaining* 05:00:27
> *ETA* 00:33 today

fraschetti, 2019-01-09:
Hi @tedder 

Can you confirm the time zone the machine running Octoprint is configured to use as well as the time zone (if any) you've configured for Octoslack to use (via Octoslack's _ETA timestamp time zone_ setting)?
It'd be helpful to see the progress output as well the output of **timedatectl** (assuming you're on linux) for that same time period.

tedder, 2019-01-09:
Reading through the code now. I'm using this in addition to PrintTimeGenius, I wonder if it is setting the estimated time to zero. I'll look at the data being returned in the API.