fraschetti/Octoslack #44: high CPU load caused by very quick reconnection attempts

hakan42, 2019-01-26:
I have just updated from 1.6.2 to 1.7.0. The 1.6.2 version is working correctly.

With 1.7.0, octoprint.log is flooded with reconnection attempts, which causes the CPU load to spike and makes printing impossible:

```
2019-01-26 13:33:48,853 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:48,893 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:48,912 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:48,913 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:48,914 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:48,914 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:48,915 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:48,945 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:49,049 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
2019-01-26 13:33:49,050 - octoprint.plugins.Octoslack - ERROR - Error Slack RTM API connection error (Exception): math range error
```

fraschetti, 2019-01-26:
Hi @hakan42 
Thanks for reporting this issue. Do you mind attaching your Octoprint.log so I can take a look at what was recorded before the errors started spewing?

For now, I'd reccomend disabling Octoslack or switching to the Webhook connection method.

Thanks!

fraschetti, 2019-01-26:
Hi @hakan42 

I've found where the issue is happening and have a bit of a fix but it's not completely obvious why that issue would have even happened in the first place. Instead of blindly submitting this fix, do you mind helping me investigate the root cause?

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

fraschetti, 2019-02-07:
Hi @hakan42 
I've made a few more changes that I believe should help - prevent infinite loops & be resistant to additional failure modes.
These changes are baking on a RasPi and Windows install on OctoPrint before I push a new release.
I'll reply here once I have more information to share.

fraschetti, 2019-02-22:
Hi @hakan42 
I've realeased 1.7.2 which contains a fix for the reconnection delay logic and some additional logic to prevent getting stuck in error/exception loops.
Please feel free to reopen this issue if you see the problem again.

closed 2019-02-22 by fraschetti