fraschetti/Octoslack #45: Failed to reconnect via Slack RTM API

JFSolution, 2019-02-06:
In the current release 1.71 I get errors in the Log:
octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM API
and I get only the octoprint service has startet message in Slack, the commands do not work.

I updatet my other octoprint Server to the new release and don't get this messages.

fraschetti, 2019-02-07:
Hi @JFSolution 

1. Can you share the details of your setup? OctoPi, custom RasberryPi build, windows?
2. Do you mind sharing a debug log that reproduces the issue? Steps below.
3. I have a number of post 1.7.1 changes related to this logic which should make it a bit more resilient. I'm baking those changes but it'd be good to see your log file to confirm your issue.

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

JFSolution, 2019-02-10:
Hi @fraschetti 
yes of course:
1. it is a custom Raspberry Pi build
2. the log (i overwrite the custom slack info with "_____"):
```
2019-02-10 11:53:40,617 - octoprint.plugins.Octoslack - DEBUG - Slack RTM Bot user id: _____
2019-02-10 11:53:40,933 - octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM API
2019-02-10 11:53:40,934 - octoprint.plugins.Octoslack - DEBUG - Attempting to connect Slack RTM API (iteration=1)
2019-02-10 11:53:40,936 - octoprint.plugins.Octoslack - DEBUG - Sleeping for 10 seconds before attempting connection
2019-02-10 11:53:51,296 - octoprint.plugins.Octoslack - DEBUG - Slack RTM API Key auth test response: {"user_id": "___", "url": "___", "team_id": "___", "user": "redocto", "team": "____", "ok": true}
```

fraschetti, 2019-02-16:
Hi @JFSolution 
Do you mind including a bit more from your log?

When starting the Slack RTM (Real Time Messaging) logic, two things happen:
1. It uses the standard HTTP REST API to validate your Slack API Token is valid (via https://slack.com/api/...) over HTTPS/port 443. Your token appears to be valid.
2. If your token is valid, it attempts to establish a WebSocket connection (https://api.slack.com/rtm) over HTTPS/port 443.

Given the initial test and the websocket connection use the same port - 443.. that shouldn't be an issue.
Unfortunately I use the official Slack Python library (https://github.com/slackapi/python-slackclient) for the WebSocket portion and their error handling and logging logic seriously lacking to say the least.

If this is still an issue for you (it hasn't fixed itself):
1. Did this ever work for you? Did this issue start with 1.7.1 or was 1.7.1 your first use of Octoslack?
2. Anything _interesting_ in your network setup? custom firewall? tcp packet shaping/QOS enabled? traffic routed through a proxy?
3. Can you validate your Pi (if a UI is installed or at least another box on your network) can use WebSockets?
https://websocketstest.com/
https://www.websocket.org/echo.html
4. Depending on your skill level, you could also try tcpdump of 443 - it wouldn't show you the unenctypted content but there may be something interesting there.

Thanks and good luck.

JFSolution, 2019-02-17:
Hi @fraschetti ,
it is still a issue for me. I use your Plugin since a few year but since a few month or so the status is not working anymore.

> * Anything _interesting_ in your network setup? custom firewall? tcp packet shaping/QOS enabled? traffic routed through a proxy?

> * Can you validate your Pi (if a UI is installed or at least another box on your network) can use WebSockets?

I have a second Octoprint which I installed with Octopi there is the status working again after the latest update. It dind't work the months before. So I don't think the router or a firewall is the problem.

> Do you mind including a bit more from your log?

2019-02-10 11:43:34,809 - octoprint.startup - INFO - ******************************************************************************
2019-02-10 11:43:34,816 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-02-10 11:43:34,817 - octoprint.startup - INFO - ******************************************************************************
2019-02-10 11:43:35,025 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-02-10 11:43:35,316 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-02-10 11:43:35,743 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-02-10 11:43:59,059 - octoprint.plugin.core - INFO - Plugin CuraEngine (<= 15.04) is disabled.
2019-02-10 11:44:00,965 - octoprint.plugin.core - INFO - Plugin Email Notifier (0.1.1) is disabled.
2019-02-10 11:44:01,154 - octoprint.plugin.core - INFO - Plugin Pushover (0.1.6) is disabled.
2019-02-10 11:44:02,862 - octoprint.plugin.core - INFO - Plugin Slack (0.2.3) is disabled.
2019-02-10 11:44:02,905 - octoprint.plugin.core - INFO - Plugin Filament Sensor Reloaded (1.0.1) is disabled.
2019-02-10 11:44:02,953 - octoprint.plugin.core - INFO - Found 28 plugin(s) providing 23 mixin implementations, 36 hook handlers
2019-02-10 11:44:02,962 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-02-10 11:44:03,033 - octoprint.server - INFO - Intermediary server started
2019-02-10 11:44:03,034 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-02-10 11:44:03,516 - octoprint.plugin.core - INFO - Found 28 plugin(s) providing 23 mixin implementations, 36 hook handlers
2019-02-10 11:44:03,581 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-02-10 11:44:04,796 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-02-10 11:44:05,227 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2019-02-10 11:44:05,409 - octoprint.plugins.multi_colors - INFO - MultiColors init
2019-02-10 11:44:10,247 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2019-02-10 11:44:14,254 - octoprint.util.pip - INFO - Version of pip is 19.0.2
2019-02-10 11:44:14,256 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-02-10 11:44:14,257 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-02-10 11:44:14,267 - octoprint.plugin.core - INFO - Initialized 23 plugin implementation(s)
2019-02-10 11:44:14,304 - octoprint.plugins.psucontrol - INFO - Running RPi.GPIO version 0.6.3
2019-02-10 11:44:14,305 - octoprint.plugins.psucontrol - INFO - Using GPIO for On/Off
2019-02-10 11:44:14,306 - octoprint.plugins.psucontrol - INFO - Configuring GPIO for pin 24
2019-02-10 11:44:14,325 - octoprint.plugin.core - INFO - 28 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
| !Email Notifier (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_emailnotifier
|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
| !Filament Sensor Reloaded (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
[... cut ...]
2019-02-10 11:53:51,297 - octoprint.plugins.Octoslack - DEBUG - Slack RTM Bot user id: U8R13JYJG
2019-02-10 11:53:51,622 - octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM API
2019-02-10 11:53:51,623 - octoprint.plugins.Octoslack - DEBUG - Attempting to connect Slack RTM API (iteration=2)
2019-02-10 11:53:51,623 - octoprint.plugins.Octoslack - DEBUG - Sleeping for 20 seconds before attempting connection
2019-02-10 11:53:57,515 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.23

fraschetti, 2019-02-21:
Hi @JFSolution 

I've made a few additional Slack RTM tweaks in 1.7.2 (just released) but I'm not entirely sure they'll solve you're issue. Give it a try when you have the time and let me know if your situation changes.

JFSolution, 2019-02-21:
Hi @fraschetti ,

I installed it and tested it, but nothing changed.
The status is still not working just the normal Octoprint Events.
Here is the new Log:

2019-02-20 12:15:59,659 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-02-20 12:15:59,671 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2019-02-20 12:15:59,671 - octoprint.server - INFO - OctoPrint 1.3.10
2019-02-20 12:15:59,678 - octoprint.plugin.core - INFO - 29 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
| !CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
| !Email Notifier (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_emailnotifier
|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
| !Filament Sensor Reloaded (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Multi Colors (1.0.16) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_multi_colors
|  Octolapse (v0.3.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse
|  Octoslack (1.7.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Printer Stats (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats
|  PSU Control JFS (0.1.5) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_psucontrol
| !Pushover (0.1.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_pushover
| !Slack (0.2.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_slack
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  TouchUI (0.3.12) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
|  Yamlpatcher (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_yamlpatcher
2019-02-20 12:15:59,707 - octoprint.environment - INFO - Detected environment is Python 2.7.9 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 900.0
|    ram: 903958528
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 2 Model B Rev 1.1
|      octopi_version: 0.13.0
|  python:
|    pip: 19.0.2
|    version: 2.7.9
|    virtualenv: /home/pi/oprint
2019-02-20 12:15:59,708 - octoprint.server - INFO - ------------------------------------------------------------------------------
2019-02-20 12:16:03,226 - octoprint.plugins.Octoslack - DEBUG - Slack RTM API Key auth test response: {"user_id": "_______", "url": "https://______.slack.com/", "team_id": "_______", "user": "redocto", "team": "_______", "ok": true}
2019-02-20 12:16:03,228 - octoprint.plugins.Octoslack - DEBUG - Slack RTM Bot user id: U8R13JYJG
2019-02-20 12:16:03,546 - octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM API
2019-02-20 12:16:03,547 - octoprint.plugins.Octoslack - DEBUG - Attempting to connect Slack RTM API (iteration=2)
2019-02-20 12:16:03,548 - octoprint.plugins.Octoslack - DEBUG - Sleeping for 20 seconds before attempting connection
2019-02-20 12:16:23,939 - octoprint.plugins.Octoslack - DEBUG - Slack RTM API Key auth test response: {"user_id": "_______", "url": "https://______.slack.com/", "team_id": "_______", "user": "redocto", "team": "_______", "ok": true}
2019-02-20 12:16:23,940 - octoprint.plugins.Octoslack - DEBUG - Slack RTM Bot user id: U8R13JYJG
2019-02-20 12:16:24,268 - octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM API
2019-02-20 12:16:24,269 - octoprint.plugins.Octoslack - DEBUG - Attempting to connect Slack RTM API (iteration=3)
2019-02-20 12:16:24,270 - octoprint.plugins.Octoslack - DEBUG - Sleeping for 40 seconds before attempting connection
[... cut ...]
2019-02-20 12:32:57,594 - octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM API
2019-02-20 12:32:57,595 - octoprint.plugins.Octoslack - DEBUG - Attempting to connect Slack RTM API (iteration=1)
2019-02-20 12:32:57,601 - octoprint.plugins.Octoslack - DEBUG - Sleeping for 10 seconds before attempting connection
2019-02-20 12:33:07,947 - octoprint.plugins.Octoslack - DEBUG - Slack RTM API Key auth test response: {"user_id": "_______", "url": "https://______.slack.com/", "team_id": "_______", "user": "redocto", "team": "_______", "ok": true}

...........

fraschetti, 2019-06-19:
Hi @JFSolution 

Just checking in to see if you've still been having Slack app/bot issues. It's been smooth sailing on my end but I wanted to check in to see if your issue had been resolved as well.

Thanks in advance.

fraschetti, 2019-07-10:
Closing this issue due to inactivity,

closed 2019-07-10 by fraschetti

JFSolution, 2019-10-25:
I reinstalled the Raspberry, now it is working without any issues.