fraschetti/Octoslack #49: minio snapshots do not show up

tedder, 2019-02-21:
I've added some debugging, but I need to figure out where minio uploads are failing. Again, this is a ticket mostly for myself, though I'd love to hear if minio is working/failing for others.

SylZap, 2019-03-06:
Hi. It work for me.
I've just edit the policy for the bucket in order to allow reading file beginning by "Snapshot".

[image]
[image]

fraschetti, 2019-06-16:
@tedder and @SylZap 

I hope this has working for you both. Just as an FYI, the snapshot images produced by Octoslack itself will have the Snapshot prefix the timelapse videos generated by Octoprint will be named differently. Octoprint uses the name of your print + a timestamp for its .mpg files.
It doesn't hurt to leave this ticket open a bit longer but if @tedder has this working, feel free to close this issue or let me know and I can close it out.

tedder, 2019-06-16:
FWIW the reason it doesn't work for me is Minio is on a private network, I think.

swills, 2021-08-29:
It works for me, but every message contains:

Snapshot error(s):

    unknown url type: ''