fraschetti/Octoslack #50: Print progress update method in-place no longer working

kazibole, 2019-04-17:
I have the print progress update method set to "in-place". It used to work correctly, however within the last few weeks I have noticed that it is no longer working (it will output a new message for each status update). Recently I have been testing OctoPrint 1.3.11 (currently on rc3), not sure if this is the cause of the problem or something unrelated.

Let me know if you need me to do any testing or provide additional information.

fraschetti, 2019-06-27:
Hi @kazibole,

I gave this a try tonight and my quick tests show the in-place update logic working as expected.

My setup:
**Connection type** Slack API Token
**Snapshot hosting** Slack
**Progress update method** In-place

If you're still seeing an issue, take a look at your Octoprint log file (/home/pi/.octoprint/logs/octoprint.log on an OctoPi deployment), specifically looking for any ERROR entries.

kazibole, 2019-07-06:
I updated to Octoslack 1.9.0 and ran some tests. The issue is still occurring for me. It appears to be related to the G-code Event that I have configured.

I have a G-code Event configured to trigger on the following serial line received by the printer:

`Recv: NORMAL MODE: Percent done: 94; print time remaining in mins: 4`

[image]

The G-code Event works as the {cmd} message does get posted to Slack (in a different channel as per my override). However when it does so, the in place updating of the print progress event does not work - it writes a new message every time. If I remove the G-code Event, the in place updating of the print progress event works as expected.

Let me know if you need any more information.

fraschetti, 2019-07-13:
Hi @kazibole 

I took a look at this recently and found the issue. The original author of that logic (I rewrote some of it but not all) added a few lines that reset the message we're updating each time a non-Progress event is processed by Octoslack. For your situation, the G-code event is being triggerd and as a result was resetting that internal logic for tracking the Slack message to update.
There were also a few other issues in that code that I found and fixed.

I'm looking into one or two other items and will push a new release once everything is ready.

kazibole, 2019-07-13:
Thanks @fraschetti! Don’t rush to release the fix just for me as I am working around the issue for now.

fraschetti, 2019-07-18:
This fix has been released with Octoslack 1.9.1

Feel free to let me know if this doesn't resolve your issue.

closed 2019-07-18 by fraschetti