fraschetti/Octoslack #51: Feature Request- automatically print completion (or any message).

diyengineer, 2019-04-22:
After print completion, how could I have the slack message automatically printout on paper with whatever printer is setup on slack? Slightly inspired by: (Where this printer will print out your tweets).
https://learn.adafruit.com/pi-thermal-printer/twitter-setup

Thanks!

diyengineer, 2019-04-22:
https://slack-bot.mobilewebprint.net/

Or.. link to HP BOT and have it print automatically instead of clicking print.

tedder, 2019-04-22:
Implementing this is getting really close to [jwz's law](http://www.catb.org/jargon/html/Z/Zawinskis-Law.html).

fraschetti, 2019-06-16:
Hi @diyengineer 

There are two ways to approach this:
1. A slack app/bot which receives the messages sent to the channel and prints them out for you. The bidirecitional websocket approach (https://api.slack.com/rtm) - feel free to poke around the Octoslack surce for "RTM"

2. A modification to Octoslack itself to execute a shell command with the event text and that shell command could print your the text for you.

#2 is actually a roadmap item for Octoslack, or at least the functionality to call out to a shell program is.

I'm going to close it this issue for now was this isn't functionality I think should be directly added to Octoslack. If something like #2 seems interesting to you, feel free to open a new issue with that enhancement request.

closed 2019-06-16 by fraschetti