fraschetti/Octoslack #52: feature request: separate "listen" toggle for readonly commands

smitty97, 2019-04-23:
I need a separate "listen to commands" toggle that only allows safe readonly commands.  So my slack buddies don't mess with prints.  :) Or, have it do stop/pause/resume only from a specified slack username.

tedder, 2019-04-23:
What commands would be remaining in this use case?

smitty97, 2019-04-23:
i suppose just 'status'

tedder, 2019-04-23:
It doesn't seem terribly worthwhile- just unset it and adjust the progress percentage to the right level.

smitty97, 2019-04-23:
Yeah, you have a point.  I'd like it to be useful though, just not open to abuse from all in that channel.  How about the allowed username list?

fraschetti, 2019-06-28:
A duplicate request - https://github.com/fraschetti/Octoslack/issues/56#issue-459986188

fraschetti, 2019-06-28:
Hi @smitty97 @tedder @Mortikhan

I've finished development on this item and it'll be included in the next Octoslack release (TBD but not far off).

What I added:
-Individual commands can be enabled/disabled completely via the Octoslack config UI
-Individual commands can be marked as 'Restricted' which if combined with a list of authorized users (this can also be supplied via the Octoslack UI), will reject commands from any unauthorized users.

I'll comment here again once the new release has gone out. In the mean time, feel free to ask any questions you might have.

Mortikhan, 2019-06-28:
Excellent news. Can't wait for the new version to come out. Thanks a ton!

fraschetti, 2019-07-10:
@smitty97 @tedder @Mortikhan 

This functionality has been relased in Octoslack 1.9.0. Feel free to provide any feedback you might have.

closed 2019-07-10 by fraschetti

Mortikhan, 2019-07-10:
Works very well. Thanks for the update and new functionality.