fraschetti/Octoslack #54: Installation problem

negativ72rus, 2019-05-19:
After installing the plugin in the list of installed it does not appear
OctoPrint: 1.3.11

fraschetti, 2019-05-19:
Hi @negativ72rus 

- Can you share the details of your setup? OctoPi, custom RasberryPi build, windows?

- Have you checked your octoprint.log? There's likely an Octoslack related error being logged there that might explain what's going on.

negativ72rus, 2019-05-19:
hi @fraschetti 
i use OctoPi

Octoslck was installed befor, but one day in log i found:
```
2019-05-14 19:23:12,225 - octoprint.plugin.core - WARNING - Could not locate plugin Octoslack
2019-05-14 19:23:12,412 - octoprint.plugin.core - ERROR - Error loading plugin anywhere
```
then i install it again
in log:
```
2019-05-14 19:56:02,092 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/fraschetti/Octoslack/archive/master.zip
2019-05-14 19:56:03,293 - octoprint.plugin.core - WARNING - Could not locate plugin Octoslack
2019-05-14 19:56:03,313 - octoprint.plugin.core - ERROR - Error loading plugin anywhere

```

fraschetti, 2019-05-25:
Hi @negativ72rus 

I've never see this problem before but it also appears you're experiencing issues with the **anywhere** plugin as well.

A few initial questions:
1. Are OS/device are you using?
2. Assuming you're on a linux distro, can you check the output of this command to ensure you can actually pull the install correctly?
curl -v -L https://github.com/fraschetti/Octoslack/archive/master.zip
3. Are there any other errors/warnings in our Octoprint log file? Either before or after the lines you shared?

ax42, 2019-05-28:
Same issue here I think -- OctoSlack working fine on an existing octopi, and am busy moving to a new odroid install.  Plugin did not restore from the backup, and installing it gives errors.

The issue seems to be missing header files for the zlib libraries - `apt-get install zlib1g-dev` fixed it for me.

```
Installing plugin "Octoslack" from https://github.com/fraschetti/Octoslack/archive/master.zip...
/home/ax/OctoPrint/venv/bin/python2 -m pip --disable-pip-version-check install https://github.com/fraschetti/Octoslack/archive/master.zip --no-cache-dir
Collecting https://github.com/fraschetti/Octoslack/archive/master.zip
Downloading https://github.com/fraschetti/Octoslack/archive/master.zip
Requirement already satisfied: OctoPrint in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.7.2) (1.3.11)
Requirement already satisfied: slackclient in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.7.2) (1.3.1)
Requirement already satisfied: slacker in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.7.2) (0.13.0)
Requirement already satisfied: tinys3 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.7.2) (0.1.12)
Requirement already satisfied: imgurpython in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.7.2) (1.1.7)
Collecting Pillow (from Octoslack==1.7.2)
Downloading https://files.pythonhosted.org/packages/81/1a/6b2971adc1bca55b9a53ed1efa372acff7e8b9913982a396f3fa046efaf8/Pillow-6.0.0.tar.gz (29.5MB)
Collecting humanize (from Octoslack==1.7.2)
Downloading https://files.pythonhosted.org/packages/8c/e0/e512e4ac6d091fc990bbe13f9e0378f34cf6eecd1c6c268c9e598dcf5bb9/humanize-0.5.1.tar.gz
Collecting minio (from Octoslack==1.7.2)
Downloading https://files.pythonhosted.org/packages/c8/bd/9fe50024059e871e65ad05385d1f0a6037308d7ccd8a302adf4699d58305/minio-4.0.17-py2.py3-none-any.whl (54kB)
Requirement already satisfied: netifaces in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.7.2) (0.10.9)
Requirement already satisfied: pytz in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.7.2) (2019.1)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (5.2.1)
Requirement already satisfied: rsa<5,>=4.0 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (4.0)
Requirement already satisfied: future<0.18,>=0.17.1 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (0.17.1)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (1.5.0.1)
Requirement already satisfied: Flask-Principal<0.5,>=0.4 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (0.4.0)
Requirement already satisfied: tornado==4.5.3 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (4.5.3)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (0.12)
Requirement already satisfied: Click<8,>=7 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (7.0)
Requirement already satisfied: futures<3.3,>=3.2 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (3.2.0)
Requirement already satisfied: werkzeug<0.16,>=0.15.1 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (0.15.4)
Requirement already satisfied: pylru<1.3,>=1.2 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (1.2.0)
Requirement already satisfied: regex!=2018.11.6 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (2019.5.25)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (0.12.2)
Requirement already satisfied: psutil<5.7,>=5.6.1 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (5.6.2)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (0.2.11)
Requirement already satisfied: monotonic<1.6,>=1.5 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (1.5)
Requirement already satisfied: typing<4,>=3.6.6 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (3.6.6)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (5.1)
Requirement already satisfied: scandir<1.11,>=1.10 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (1.10.0)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (1.6.5)
Requirement already satisfied: markdown<3.1,>=3.0 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (3.0.1)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/ax/OctoPrint/venv/lib/python2.7/site-packages (from OctoPrint->Octoslack==1.7.2) (0.7.19)
[... cut ...]
https://pillow.readthedocs.io/en/latest/installation.html
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-bIa9Iu/Pillow/setup.py", line 804, in <module>
raise RequiredDependencyException(msg)
__main__.RequiredDependencyException:
The headers or library files could not be found for zlib,
a required dependency when compiling Pillow from source.
Please see the install instructions at:
https://pillow.readthedocs.io/en/latest/installation.html
----------------------------------------
```

fraschetti, 2019-06-16:
For reference, the same issue was reported for Octolapse as well:
https://github.com/FormerLurker/Octolapse/issues/342

fraschetti, 2019-06-16:
Closing this issue as this isn't specifically an Octoslack issue and a suitable workaround has been identified:

`apt-get install zlib1g-dev`

closed 2019-06-16 by fraschetti