fraschetti/Octoslack #55: Updated to 1.8.0, settings not being saved or shown properly

kazibole, 2019-06-18:
Upgraded from 1.7.2 to 1.8.0. I noticed that my custom gcode events were no longer working. Checked the settings page and they were gone. I can re-create them but when I restart OctoPrint they will be lost again. I also noticed that some of the other settings are behaving strangely.

ETA timestamp time zone is blank: 
[image]

Snapshot hosting is also blank (it should be "Slack"): 
[image]

fraschetti, 2019-06-19:
Hi @kazibole 

I've identified and fixed the issue - you may need to resave your settings once you upgrade.

I've just pushed 1.8.1 - feel free to give it a try and let me know whether or not I've resolved your issue.

Thanks in advamce.

kazibole, 2019-06-19:
Appears to be fixed in 1.8.1!

closed 2019-06-19 by kazibole