fraschetti/Octoslack #57: requires python 3.6 error

diyengineer, 2019-07-18:
Hello,

I had version 1.9 working on a machine, and went to install it on another but it was giving me a new error that python 3.6 was not installed. Is this a bug, or am i doing something wrong? Thank you!

Installing plugin "Octoslack" from https://github.com/fraschetti/Octoslack/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/fraschetti/Octoslack/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/fraschetti/Octoslack/archive/master.zip
Downloading https://github.com/fraschetti/Octoslack/archive/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Octoslack==1.9.0) (1.3.11)
Collecting slackclient (from Octoslack==1.9.0)
Downloading https://www.piwheels.org/simple/slackclient/slackclient-2.1.0-py2.py3-none-any.whl (51kB)
**_slackclient requires Python '>=3.6.0' but the running Python is 2.7.13
Error!_**
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

fraschetti, 2019-07-18:
Hi @diyengineer 

I've just released Octoslack 1.9.1 to fix this issue.
Long story short, OctoPrint only operates on Python 2 but Slack's latest client library only supports Python 3. Octoslack will stay on the older stream of their client library to work around this issue. 

Please let me know if this doesn't resolve your issue.

fraschetti, 2019-07-24:
This issue should be resolved. Closing this issue (feel free to reopen it if you still see a problem)

closed 2019-07-24 by fraschetti