fraschetti/Octoslack #58: [Feature Request] Estimated print time for new print notification

Billiam, 2019-07-18:
I'm not sure what the right approach to this is, but: 

For new prints, my notifications rarely if ever display the estimated print time, unless I'm reprinting gcode which has been analyzed earlier.

For most prints, my octoslack notifications show "Estimated print time: N/A". If I enable notifications for metadata analysis complete, I get notifications for every uploaded file, even if they're not going to be printed, and (probably because I'm using the PrintTimeGenius plugin), I get two of these notifications back to back.

Instead, it would be nice if the existing print started notification could be updated using the slack api once an updated estimate was available.

Not sure if there's a better approach though, any suggestions?

fraschetti, 2020-05-24:
Hi @Billiam 

Do you still see these issues? If so, let me know and we'll kick off a discussion to see what can be done.

floslo, 2020-09-25:
I see these issues with "Print started" on Prusa MK3S printers.

ETA availability depends on print start method.
1. Drag file to OctoPrint web UI and start print there: ETA shown.
2. PrusaSlicer [send to OctoPrint](https://help.prusa3d.com/en/article/sending-files-to-octoprint-duet_1663) feature: Most often "ETA N/A".
3. Start print from SD-card via Prusa printer LCD menu: ETA never shown.
  
**2. PrusaSlicer send to OctoPrint feature**

Here prints start before file metadata analysis finish.  

To reproduce that enable Octoslack notifications for `Print started` [PS], `File analysis started` [FAS] and `File analysis finished` [FAF] and then print from PrusaSlicer.  

The result depends on the OctoPrint [analyze gcode](https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/templates/dialogs/settings/features.jinja2#L61) setting:

- If set to `only when idle (not when printing)` we get notifications FAS followed by PS with `ETA N/A`. No FAF.
- If set to `always` we get FAS, PS with `ETA N/A` and finally also FAF with correct ETA.

This FAF workaround is not ideal. Because a FAF notification shows every time we drag a file into OctoPrint web UI, even if  we plan to start the print at a later time or decide not to print it.

Better workaround: Set a custom Octoslack notification for GCODE line  [G80](https://reprap.org/wiki/G-code#G80:_Mesh-based_Z_probe) ([mesh bed leveling](https://help.prusa3d.com/en/article/mesh-bed-leveling_112163) start). Because G80 always runs on Prusa MK3S, runs only once and by that time file analysis has had time to finish so the notification shows ETA.

I suppose a real fix would delay print start until file analysis finish? Which might also require changes to PrusaSlicer and/or OctoPrint?

**3. Start print from SD-card via Prusa printer LCD menu**  
I have found no workarounds here. About a minute into the print the OctoPrint UI updates to show (shortened) print file name and duration. But I find no way to have that change trigger an Octoslack notification.

 This might also require changes to OctoPrint?