fraschetti/Octoslack #59: Slack instructions are unclear

timur-tabi, 2019-07-24:
The settings page for Octoslack says that I need to specify an API Token, and then it says I can get that via "Apps & integrations --> Manage --> Custom Integrations --> Bots".  I cannot find the "Apps & Integrations" menu item.  Can you please provide a detailed walkthrough for enabling Slack support?

timur-tabi, 2019-07-24:
I managed to create an API Token by visiting https://api.slack.com/custom-integrations/legacy-tokens (you should probably update the instructions).  However, even with the Token, I don't know how to communicate with OctoSlack from my Slack session.  It talks about "@botname", but I have no idea what that is supposed to be.

timur-tabi, 2019-07-24:
Ok, somehow I managed to get OctoSlack to post messages to a Slack channel stating when prints have started.  However, I can't figure out how to send messages to OctoSlack.  I'm primarily interested in getting occasional snapshots of my prints.

[image]

timur-tabi, 2019-07-25:
Is it possible for me to use OctoSlack to send instructions to the printer?  For example, if I see that my print has failed, can I use OctoSlack to tell the print to cancel the print?

fraschetti, 2019-07-26:
> The settings page for Octoslack says that I need to specify an API Token, and then it says I can get that via "Apps & integrations --> Manage --> Custom Integrations --> Bots". I cannot find the [...]

Hi @timur-tabi 
You're correct that the Slack UI has changed slightly but ultimately you're still looking for the "Custom Integrations --> Bots" UI (see the screenshots below). I'll make a note to update the instructions if I can think of a better way to describe how to get to that UI.

[image]
[image]

fraschetti, 2019-07-26:
> I managed to create an API Token by visiting https://api.slack.com/custom-integrations/legacy-tokens (you should probably update the instructions). However, even with the Token, I don't know how to [...]

Hi @timur-tabi ,

You're correct that Slack has flagged the current Custom Integrations (Bots and WebHooks) to be deprecated at some point. For the moment they should still function just fine. As for how to create your app integratin, you want to create a new Bot. The screen to create a new bot (included below) will allow you to specify the botName you want to use. I always use "octoslack" which allows me to then use  **@octoslack status**  and other supported commands.

[image]

fraschetti, 2019-07-26:
> Ok, somehow I managed to get OctoSlack to post messages to a Slack channel stating when prints have started. However, I can't figure out how to send messages to OctoSlack. I'm primarily interested in getting occasional snapshots of my prints.
> 
> [image]

Hi @timur-tabi ,

As per my previous comment, you can easily ask Octoslack for a status update whenever you like (e.g. **@octoslack status**) but most people simply configure the Progress event to send periodic updates (% intervals, print height intervals, or time intervals).

[image]

fraschetti, 2019-07-26:
> Is it possible for me to use OctoSlack to send instructions to the printer? For example, if I see that my print has failed, can I use OctoSlack to tell the print to cancel the print?

Hi @timur-tabi 

Once you have your bot setup correctly and observe a print issue, you will be able to use Slack to cancel the print - **@octoslack stop**
Unfortunately this is currently a manual process (observe a problem, issue the command to cancel the print); feel free to submit an enhancement request if you have an idea for how to integrate this flow automatically.

In addition to the events Octoslack can respond to, when using the Slack bot configuration as you appear to be (not available when only using a WebHook):
[image]

PrinterFarm3000, 2019-07-28:
Hi Fraschetti,

Would it be possible to send a "print" command in Slack to Octoprint? (assuming the file was already loaded in octo, the printer would then begin the job. Or, if you could "load and print" a specific file from the printer... This would be extremely helpful)

Thank you

PrinterFarm3000, 2019-07-28:
Also, with multiple printers, (say they are named "1,2,3,4,5,") is there a command in slack to address all of these printers at once? Instead of manually addressing them @1 status @2 status @3 status @4 status @5 status - Is there a @ALL status command that you know of? I have attempted with a couple commands already but it didn't work for me. Thanks!

fraschetti, 2019-07-28:
Hi @timur-tabi 

There is another plugin (not for Slack) that has the concept of "list" and "print X" but those aren't capabilites I've thus far been asked to add to Octoslack. For the most part I developer what I need and then what the consumers of the plugin request.
I'd suggest opening a new issue specfic to the request of being able to start a print from Slack.

-Chris

fraschetti, 2019-07-28:
Hi @PrinterFarm3000 

The way the Slack API integration works is that the plugin actually receivees **all** messages sent to the channel. It uses the Slack bot's username (e.g. **@1**) to determine if the message was intended for the bot. It's concievable (and not a lot of effort) to allow for another special keyword (shared across all your printers/Octoslack instances) to would serve the same purpose.
I'd go ahead and open a new issue for this functionality and I'll add it in as soon as time permits.

-Chris

PrinterFarm3000, 2019-07-28:
Thanks Chris, I'll write a request- I think a lot of people would find these features useful!

fraschetti, 2020-05-24:
Closing out this issue as I suspec @timur-tabi has figured out his setup steps - the new Slack App flow is also a bit different so we'll see if we need to revisit this in the future.

@PrinterFarm3000 has also opened https://github.com/fraschetti/Octoslack/issues/60 so we'll address that request to start prints from Slack there.

closed 2020-05-24 by fraschetti