fraschetti/Octoslack #61: Doesn't respond to commands

GaryBoone, 2019-07-28:
I've set up OctoSlack successfully enough to see the 'successfully connected to printer' messages and print updates from the printer. A print run sends progress photos every 15 minutes as I configured it to. However, I can't send commands such as help or status and receive a reply.

In the OctoSlack settings, I have checked "Listen for commands" and checked all of the commands. I have not checked the "Restricted" boxes or entered any "Authorized users." I have "Use Slack identity" checked.

In the mobile Slack app, in the channel where the OctoSlack printer updates appear, I'm typing "@NNN help" and "@NNN status" where NNN is the name that appears in Slack for the OctoSlack APP.

So I think I've configured it according to the documentation. How can I figure out what's wrong?

fraschetti, 2019-08-01:
Hi @GaryBoone 

Please see the steps below to enable the Octoslack logging for the octoprint.log file. Once enabled, we should have a bit more information. The log file will write out the bot name (just to confirm), the bot's internal ID (which is what shows up in the messages it receives) as well as any related errors.
It's also worth pointing out that the Slack API connection uses WebSockets; if you have anything unique in your network setup, that could also be the cause.

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

fraschetti, 2019-08-17:
Hi @GaryBoone 

Please let me know if you still need help with your issue.

GaryBoone, 2019-08-18:
I see a series of messages like:

> 2019-08-18 13:10:35,835 - octoprint.plugins.Octoslack - DEBUG - Sleeping for 5 seconds before attempting connection
2019-08-18 13:10:41,291 - octoprint.plugins.Octoslack - DEBUG - Slack RTM API Key auth test response: {"user_id": "UUUU", "url": "https://xxxx.slack.com/", "team_id": "TTTT", "user": "uuuu", "team": "tttt", "ok": true}
2019-08-18 13:10:41,292 - octoprint.plugins.Octoslack - DEBUG - Slack RTM Bot user id: UUUU
2019-08-18 13:10:41,874 - octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM API
2019-08-18 13:10:41,877 - octoprint.plugins.Octoslack - DEBUG - Attempting to connect Slack RTM API (iteration=1)

But also the successful updates like:

> 2019-08-18 13:07:19,265 - octoprint.plugins.Octoslack - DEBUG - Slack API upload snapshot response: {"ok":true,"file":{"id":"XXXXX","created"
:1566158836,"timestamp":1566158836,"name":"snapshot.jpg","title":"Progress snapshot taken @ 45%","mimetype":"image\/jpeg","filetype":"jpg",
...
2019-08-18 13:07:19,266 - octoprint.plugins.Octoslack - DEBUG - Uploaded asset to Slack in 3.36 seconds
2019-08-18 13:07:19,269 - octoprint.plugins.Octoslack - DEBUG - Deleting local Slack asset: /tmp/Snapshot_c5b980dcc1f311e99c62b827ebe4bbb9.png

In Slack, I see progress updates with photos every 15 minutes. I next tried typing 'info' and '@botname info' into the Slack channel, but nothing appeared in the Octoslack logs.

fraschetti, 2019-08-19:
Hi @GaryBoone 

Can you share the output of the following commands:

pip show websocket_client
pip show slackclient

I can't explain it just yet but your client is obviously able to make the HTTP/S calls to Slack but is then failing somewhere within the Slack client's logic to establish a WebSocket connection. A WebSocket starts over the same HTTP/S protocol that your low obviously shows but is then upgraded to a proper WebSocket. Any chance you have firewall rules that might be blocking this connection? The ports should still be the same but perhaps there's something unique in your network that might be contributing to this problem.

If a solution doesn't present itself, I'll likely work up a test python app to see if we can further diagnose the issue.

GaryBoone, 2019-08-20:
```
$ pip show websocket_client
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
$ pip show slackclient
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
$
```

The network is a vanilla Ubiquity setup. So there could be a firewall default.

mika1337, 2020-04-24:
Hi,

I am having the same issue and was wondering if this was a scope related issue. I enabled `app_mentions:read` and `commands` without success, can you confirm which scope is required for this feature ?

Also is there a document which gives the required/recommended slack scopes for this plugin to work properly ?

mika1337, 2020-04-24:
After further investigation my issue seems related to RTM:
```
2020-04-24 14:25:02,812 - octoprint.plugins.Octoslack - DEBUG - Slack RTM (v1) API Key auth test response: {"user_id": "XXXXX", "url": "https://XXXXX.slack.com/", "team_id": "XXXXX", "user": "octoslack", "team": "XXXXX", "ok": true, "bot_id": "XXXXX"}
2020-04-24 14:25:02,813 - octoprint.plugins.Octoslack - DEBUG - Slack RTM (v1) Bot user id: U012FRWBPPC
2020-04-24 14:25:03,142 - slackclient.client - WARNING - Failed RTM connect
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/slackclient/client.py", line 140, in rtm_connect
    self.server.rtm_connect(use_rtm_start=with_team_state, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/slackclient/server.py", line 168, in rtm_connect
    raise SlackLoginError(reply=reply)
SlackLoginError
2020-04-24 14:25:03,143 - octoprint.plugins.Octoslack - ERROR - Failed to reconnect via Slack RTM (v1) API
2020-04-24 14:25:03,144 - octoprint.plugins.Octoslack - DEBUG - Attempting to connect Slack RTM (v1) API (iteration=1)
2020-04-24 14:25:03,145 - octoprint.plugins.Octoslack - DEBUG - Sleeping for 10 seconds before attempting Slack RTM (v1) connection
```

I guess that OctoSlack relies on RTM to answer commands, and from the Slack API documentation it seems that RTM is being deprecated.
I'll give it a try creating a "Classic Slack App" which supports RTM.

If anyone can confirm that RTM is needed for handling commands ?

fraschetti, 2020-04-25:
Hi @GaryBoone and @mika1337 
When you created your Slack app, do you recall if it was a new or classic app? Easiest way to know is to check if your app is rqeuired to be configured with "Scopes" (e.g. chat:write) or not. New apps require individual Scopes (permissions) to be applied whereas the classic ones do not.

The _current_ version of Octoslack utilizes RTM to listen to the channel messages and respond to commands. Unfortunately RTM (a persistent WebSocket to Slack) is only supported by the classic apps/bots and the new apps require different logic. 

The good news is that I'm working on adding support for the new Slack apps so if this is your issue, stay tuned for the next release.

@mika1337 - I'll almost certainly be documenting the various scopes the plugin will need to work; you can expect that info once the next version comes out,

mika1337, 2020-04-25:
@fraschetti the first token I used (which didn’t work) was a new app. I had messages and images sent from OctoSlack working, but commands didn’t work (as excepted since RTM is not available for new apps)
I’ll be waiting for the next release, btw thanks for your great work.

fraschetti, 2020-05-24:
Hi @GaryBoone and @mika1337 

I've just released Octoslack 2.0.0 which has a lot of related changes relevant to this conversation.
-New Slack App Bot support using the Slack Web API instead of Slack RTM (it now polls an HTTP endpoint instead of using a persistent WebSocket).
Relevant to what we were discussing here...

- For anyone using new Slack Apps/Tokens, you'll be forced to use the Web API which is a bit more reliable than the RTM/WebSocket but is also slower/delayed (we have to poll within the limits of what Slack will allow)

- For anyone using an older/Classic Slack App/Token, I've assed a 'Classic' option which will allow people to revert to the RTM method although I expect Slack to completely remove support for RTM at some point. For people who do revert to the RTM method, that logic has also been reworked a bit in hopes that it's a bit more reliable.

The Octoslack settings UI should be seen as the source of truth for the App Scopes required by the new Slack Apps but here's a snapshot of the current scopes that may be required depending on what functioanlity you're using:

### **Slack Bot Webhook Scopes**
Bot Token Scopes requied for posting to a channel
- channels:read
- chat:write
- groups:read

Bot Token Scopes requied for uploading assets/files to Slack
- files:write

### **Slack Bot Access Token Scopes**

Bot Token Scopes requied for posting to a channel
- channels:read
- chat:write
- groups:read

Bot Token Scopes requied for uploading assets/files to Slack
- files:write

Bot Token Scopes requied for listening for and responding to commands
- channels:history
- groups:history
- reactions:write
- users:read

Good luck and feel free to reopen this issue or create a new one if you have issues going forward.

closed 2020-05-24 by fraschetti

mika1337, 2020-06-03:
Hi @fraschetti, just upgraded to version 2.0.0 and moved to new bot API with permissions, everything seems to work well including commands.
Thanks !