fraschetti/Octoslack #62: Nothing received in Riot.im

thearduinoguy, 2019-08-17:
Set everything up according to the instructions and yet I am not getting any messages in Riot.im 

Please help.

fraschetti, 2019-08-17:
Hi @thearduinoguy 

First steps would be to provide the octoprintlog file. Steps below.

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

lcalaresu, 2019-11-27:
Hello there, 

I had the same problem with an old version of Matrix (**1.3.1**) that was released on 17th October (same date as this issue). However, everything works fine with the latest version of Matrix (**1.6.0**) released yesterday... I believe this issue can be closed.

I encourage those having the same issue to **upgrade their Matrix server** (if possible).

Here is the error with the old version :
```
2019-11-27 16:29:17,639 - octoprint.plugins.Octoslack - DEBUG - Send Matrix msg start
2019-11-27 16:29:18,007 - octoprint.plugins.Octoslack - ERROR - Matrix send error: 401: {"errcode":"M_UNKNOWN_TOKEN","error":"Invalid macaroon passed.","soft_logout":false}
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 3336, in send_slack_message
    user_id=matrixUserID,
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matrix_client/client.py", line 152, in __init__
    self._sync()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matrix_client/client.py", line 555, in _sync
    response = self.api.sync(self.sync_token, timeout_ms, filter=self.sync_filter)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matrix_client/api.py", line 105, in sync
    api_path=MATRIX_V2_API_PATH)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/matrix_client/api.py", line 691, in _send
    code=response.status_code, content=response.text
MatrixRequestError: 401: {"errcode":"M_UNKNOWN_TOKEN","error":"Invalid macaroon passed.","soft_logout":false}
```

fraschetti, 2019-12-22:
Closing this issue per as it has been idle for quite a while.
1.9.5 also fixed a potential issue with Matrix/Riot.im (a typo on the server URL) that I must have introduced at some point.

closed 2019-12-22 by fraschetti