fraschetti/Octoslack #63: Matrix/Riot : Emoji code and Markdown formatting are not interpreted ?

Auroch, 2019-08-19:
Hello, I have just configured the plugin and I notice that the emoji codes already proposed in the messages (or others) are not interpreted in the chat rooms.

I have already noticed that in the Riot client, you have to choose the emoji from a list while typing your text, if you leave the code :dog:, it is not interpreted, so it may be a problem of the same order, or even a design difficulty in Matrix. But since there may be something to do on the plugin side, I prefer to report it.

On the other side, the MarkDown code is correctly interpreted in the Riot client, once the message is sent, bold, italics, etc. , it 'll appear correctly. But not if we use them in Octoslack... an idea of what could cause this?

Here, no rush, just the desire to bring up a detail to make this fantastic plugin even more attractive: -)

nordurljosahvida, 2020-08-14:
@Auroch OT: how did you specify matrix room? `!id:domain` or `#id:domain`? thanks

dwmunster, 2020-11-21:
I am also experiencing this issue and available to test. 

The issue is that messages are sent as rendered HTML (https://github.com/fraschetti/Octoslack/blob/master/octoprint_Octoslack/__init__.py#L4349). Thus, any markdown/emoji must be processed on the Octoslack side if you want them to display properly. Sending the [html symbol](https://www.w3schools.com/charsets/ref_emoji.asp) (e.g. `&#10134;`) works just fine.