fraschetti/Octoslack #65: configuration question re multiple printers

rgordon, 2019-10-13:
I just discovered this plug-in, and have it already configured to send messages to a test channel. 
i'm using webhook (attached to a Slack app instead of the old way, but same result.)
i'm looking at hooking up several different printers, each on their own OctoPi server. 
And that leads to my questions.
Q1:i was trying to change some of the notification messages, but am not seeing the changes posted to Slack channel. Just the default text is posted. Is there some on-disk location that needs to be alterered? I was using the form available in the plug-in manager, w/o luck.
Q2- does "channel override" enable me to override the channel associated with the webhook or does it have some other purpose?
Q3 - the "enable system commands" - is that basically a hook to allow one to execute scripts on the local disk?

fraschetti, 2019-10-13:
Hi @rgordon 
I've supplied a few answers below. Good luck!

> Q1:i was trying to change some of the notification messages, but am not seeing the changes posted to Slack channel. Just the default text is posted. Is there some on-disk location that needs to be alterered? I was using the form available in the plug-in manager, w/o luck.
Are you changing the Message or the Fallback? The Fallback text is typically used for notification systems (iPhone/Android lock screens, Windows 10 notification system, etc.) where it wouldn't make sense to try render the full message text. As far as making updates is concerned, your updates should be recognized immediately; for the settings that do require a restart, the plugin will prompt you. Any settings changes will be reflected in the Octoprint config file. Note: Only if you've changed the valuef from something other than the default will it show up it in the yml file.
/home/pi/.octoprint/config.yaml

> Q2- does "channel override" enable me to override the channel associated with the webhook or does it have some other purpose?
Yes. You can specify one or more channels you'd like to send messages to instead of the default channel for your webhook.

> Q3 - the "enable system commands" - is that basically a hook to allow one to execute scripts on the local disk?
Yes, exactly.

rgordon, 2019-10-28:
thanks. I see the messages showing correctly now. I'm wishing that there was a "printer-name" macro to go along with all of the others - I ended up adding each one manually.
but - while doing so on the first printer, I've been trying to get the notifications to sent to some other channel than the default for the webhook. so far, I set the one at the top - its had no effect. And I set for the channel override in one of the messages - it had no effect either. Is there a specific syntax other than simply the channel name? (I'm trying to setup several printers in a lab, each going to their own channel so that people can monitor their own jobs.)

fraschetti, 2019-10-28:
Hi @rgordon 

Can you describe your "printer-name macro" a bit further?
Are you looking for the Slack messages to include a unique label per printer (if so, where were you hoping it would be displayed?) or are you using the Slack API (Slack bot) and would like to use different @bot1 @bot2 @bot3, etc. per printer to run commands?

rgordon, 2019-11-04:
for the printer-name: either something settable or perhaps derived from the hostname.
what I'm trying to do is support multiple printers - one Octoprint server per printer.
What i'm trying to determine as well is the best way - whether I can have a single Slack App, with 1 or multiple WebHooks - so that in the end each printer can send traffic to its own #channel. I thought I could do each printer server with the same web hook and let it determine the channel. That hasn't worked out for me yet - although admittedly i haven't had a chance to test it out further this past week - maybe this coming week I will.  If that doesn't work I suppose I'll create a separate WebHook URL (or even a separate Slack App if multiple hooks per app are not allowed) for each printer - each defaulting to a separate channel.

rgordon, 2019-11-08:
Testing this out again tonight,  I have not been able to get it to send to anything other than the default channel for the webhook account. I've set the channel-override value at the top of the settings, both with and without leading '#'. I also tried setting it at the individual commands level. no dice. about to give up and just create a separate webhook for each printer. unless someone has better ideas...

fraschetti, 2019-11-08:
Hi @rgordon 

You might try enabling logging in hopes the response from Slack will contain the answer.

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

rgordon, 2019-11-08:
thanks, i'll try to take a look at it more this weekend. it could well be on the Slack side of course. i was poking around in there last night looking for some clues as well.

rgordon, 2019-11-10:
Enabled the log and took a look - it *seems* to be grabbing the correct channel out of the settings, and posting to Slack with it w/o error. so - I'm guessing its something on the Slack side.
There, I've created a full-fledged "app" per their current guidance, and enabled a webhook url on that.

2019-11-10 22:55:39,499 - octoprint.plugins.Octoslack - DEBUG - Event: Connected, NotificationEnabled: True, CommandEnabled: False, Payload: {'baudrate': 0, 'port': u'AUTO'}
2019-11-10 22:55:39,503 - octoprint.plugins.Octoslack - DEBUG - Printer data: {'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeOrigin': None}, 'state': {'text': 'Operational', 'flags': {'cancelling': False, 'paused': False, 'operational': True, 'pausing': False, 'printing': False, 'resuming': False, 'sdReady': False, 'error': False, 'ready': True, 'finishing': False, 'closedOrError': False}}, 'currentZ': None, 'job': {'file': {'date': None, 'origin': None, 'size': None, 'name': None, 'path': None}, 'estimatedPrintTime': None, 'user': None, 'filament': {'volume': None, 'length': None}, 'lastPrintTime': None}, 'offsets': {}}
2019-11-10 22:55:39,585 - octoprint.plugins.Octoslack - DEBUG - Octoslack connection method: WEBHOOK
2019-11-10 22:55:39,588 - octoprint.plugins.Octoslack - DEBUG - postMessage - Channels: #eta-printmon-prusa-04, JSON: [{"footer": "Printer: Operational, RasPi: 47.8\u00b0C", "color": "good", "fields": [], "mrkdwn_in": ["text", "pretext"], "pretext": ":heavy_minus_sign:  Successfully connected to printer (eta-octopi-04) :computer:", "fallback": "Successfully connected to printer"}]
2019-11-10 22:55:39,591 - octoprint.plugins.Octoslack - DEBUG - Slack WebHook postMessage json: {"as_user": false, "attachments": [{"footer": "Printer: Operational, RasPi: 47.8\u00b0C", "color": "good", "fields": [], "mrkdwn_in": ["text", "pretext"], "pretext": ":heavy_minus_sign:  Successfully connected to printer (eta-octopi-04) :computer:", "fallback": "Successfully connected to printer"}], "channel": "#eta-printmon-prusa-04"}
2019-11-10 22:55:39,612 - octoprint.plugins.ipOnConnect - INFO - ipOnConnectPlugin: 10.150.150.96
2019-11-10 22:55:40,095 - octoprint.plugins.Octoslack - DEBUG - Slack WebHook postMessage response: ok

fraschetti, 2019-12-22:
Hi @rgordon 

FYI that in 1.9.5, when using the Slack API Token (bot/web socket connection) opting to not use the default slack identity and providing an alternate username will allow allow for printer/OctoPrint instance specific commands.

e.g. share one API Token but addres each printer individually:
@printer1 status
@printer2 status

but I also left the listener in for the original bot username. Issuing commands to that original username will instruct each printer to also report its status - so you can communicate with one printer or all, it's up to you.

fraschetti, 2019-12-22:
Please let me know if you have any outstanding asks or issues (we might want to create unique git issues per item).

Thanks!

rgordon, 2019-12-23:
you know, i haven't had much time to work on this thing lately but will try to in the next couple of weeks. I hadn't planned on implementing a bot interface, using the API token, but may need to reconsider that.

one more question - is it possible to extend the events the plugin responds to easily? one of the use cases is for that darned Prusa MMU2 which demands attention way too often.
it sends out special "MMU2 Needs Attention" messages and pauses.

fraschetti, 2022-11-08:
Better late than never here.
This conversation is a few years old so I'm going to resolve it. Feel free to communicate back if needed.

As for the MMU2 question, if you'd still like to discuss this, feel free to open another ticket and we can focus on that topic there. Thanks!

closed 2022-11-08 by fraschetti