fraschetti/Octoslack #66: [Feature Request] Print User who started the print

JFSolution, 2019-10-25:
show the user who startet the print in the slack message
and maybe send the messages only to the user who startet the print

fraschetti, 2020-04-18:
@JFSolution 

Which user are you interested to expose?

1. The Slack user who started a print via a command issued from Slack
2. The OctoPrint user who started the print via the OctoPrint UI

fraschetti, 2020-04-18:
As for sending messages only to the user who started the print. For Slack, it's easiest to use a dedicated channel. I've gone down the DM path for Slack before and it unfortunately ended in failure. The plugin can do this in other chat platforms but Slack itself has some limitations that I previously hit.

nebbles, 2020-04-20:
I've never used option 1 above, but I assume that when initiated from Slack, it will be linked via a specific account in Octoprint.

I would recommend using the Ocotoprint user as described in the UI. It's relatively trivial (as we have done in current start-up) of giving people each their own Octoprint account. That way we could clearly see who's print is currently running (whether in the Octoprint UI or, hopefully 🤞, in the Slack status updates).

JFSolution, 2020-04-20:
The octoprint user which started the print.
I made a fork and added the feature.
https://github.com/JFSolution/Octoslack
To post only to the user who started the print. There should be a way to associate the slack- with the octoprint-user.

fraschetti, 2020-05-24:
Hi @JFSolution 

I used your changes as the base of what I eventually ended up integrating into Octoslack.
I've made an attempt in the just relased Octoslack 2.0.0 to satisfy both camps - those who want to see the OctoPrint username and those who want to see the Slack username.

The plugin will include these values where it makes sense (e.g. list the Slack username of the user who cancelled a print via Slack) but they're also available via the new {octoprint_user} and {slack_user} message variables.

FYI @nebbles ^^

Please let me know if these new changes don't satisfy your needs.

closed 2020-05-24 by fraschetti