fraschetti/Octoslack #68: [Feature Request] Microsoft Teams Integration

shinister, 2019-11-04:
migrating from slack to microsoft teams, would be very nice to have Octoslack post status updates (with imgur snapshots) to my group's 3D printing channel on teams via the avaiable incoming webhook connector.

shinister, 2019-11-04:
...trying to avoid needing a third party slack/teams channel integration, Octoslack is the main use case we have that would require it.

fraschetti, 2020-04-16:
Hi @shinister 

FYI - I've just pushed changes to support Microsoft Teams. This will come along with next Octoslack release.

shinister, 2020-04-24:
much appreciated, @fraschetti !

will keep my eye out for the next release.

fraschetti, 2020-05-24:
Hi @shinister 

I've just released Octoslack 2.0.0 which includes Teams support. Let me know if you run into any issues. Quick FYI - Teams was likely the most limiting platform I've integrated with to date in terms of how they handle attachments/image uploads. I needed to add some extra logic to resize snapshots to fit within what Teams would allow in order for the snapshot image to actually render. 

Good luck!

closed 2020-05-24 by fraschetti