fraschetti/Octoslack #69: Printer reports print progress as N/A at Print Progress Interval when no print is active

dgsmithmn, 2019-11-05:
My printer is reporting ":heavy_minus_sign: Print progress N/A :horse_racing:" every 30 minutes (what I have the Interval - Time (minutes) set to in Print Progress). I do not have Printer status heartbeat notifications enabled.

fraschetti, 2019-11-08:
Hi @dgsmithmn 

Please see the steps below to enable the Octoslack logging for the octoprint.log file. Once enabled, we should have a bit more information. Octoslack only responds to OctoPrint events so we'll see which events your instance is responding to.

Enable DEBUG logging
Reproduce the issue
Send Octoprint.log
Instructions to enable DEBUG logging for Octoslack

Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml (or appropriate path for your OctoPrint installation)
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
Restart OctoPrint
sudo service octoprint restart
For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

dgsmithmn, 2019-11-13:
Here's a portion of what the log is reporting:

2019-11-13 06:28:23,164 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 06:28:45,228 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1771222}
2019-11-13 06:43:23,167 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 06:43:45,248 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1772122}
2019-11-13 06:58:23,170 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 06:58:45,241 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1773022}
2019-11-13 07:13:23,173 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 07:13:45,213 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1773922}
2019-11-13 07:28:23,175 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 07:28:45,223 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1774822}
2019-11-13 07:36:13,771 - octoprint.plugins.Octoslack - ERROR - Slack RTM API read error (Exception): Unable to send due to closed RTM websocket
2019-11-13 07:43:23,178 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 07:43:45,217 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1775722}
2019-11-13 07:58:23,180 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 07:58:45,248 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1776622}
2019-11-13 08:13:23,183 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-11-13 08:13:45,254 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 1777522}
2019-11-13 08:28:23,186 - octoprint.server.heartbeat - INFO - Server heartbeat <3

The Octoslack plugin is reporting the progress every half hour at 6:32, 7:02, 7:32 etc, but it is always at 100% since the print is finished. 

[image]

dgsmithmn, 2019-11-13:
Here are my Print Progress settings
[image]
[image]
[image]
[image]

fraschetti, 2019-12-19:
Hi @dgsmithmn 

I've been unable to reproduce this thus far. If you're still experiencing a problem, can you follow my steps above to enable DEBUG logging for this plugin and then report back with a more detailed log dump?

Thanks!

fraschetti, 2020-04-18:
Hi @dgsmithmn 

Let me know if you are still experiencing the issue and if you can provide a debug log for me to take a look at.

fraschetti, 2020-05-24:
Hi @dgsmithmn 
It's been six months since you reported your issue and I wasn't able to reproduce the issue in my limted testing. If this is still a problem for you, feel free to reopen this ticket with the info I requested above and we'll go from there.

closed 2020-05-24 by fraschetti