fraschetti/Octoslack #70: Error during "pillow" dependency installation

rafalmag, 2019-11-05:
I tried to install the plugin and got following error during "pillow" installation:

```
Installing plugin "Octoslack" from https://github.com/fraschetti/Octoslack/archive/master.zip...
/home/octoprint/OctoPrint/venv/bin/python -m pip --disable-pip-version-check install https://github.com/fraschetti/Octoslack/archive/master.zip --no-cache-dir
Collecting https://github.com/fraschetti/Octoslack/archive/master.zip
Downloading https://github.com/fraschetti/Octoslack/archive/master.zip
Requirement already satisfied: OctoPrint in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages (from Octoslack==1.9.1) (1.3.12)
Collecting slackclient<2.0.0
Downloading https://files.pythonhosted.org/packages/7c/2f/9b6db4c7686be3dc6b4244eaef4f37226a2ccca1a796516d4adcdfeefe56/slackclient-1.3.2.tar.gz
Collecting slacker
Downloading https://files.pythonhosted.org/packages/cc/71/7ef8a02070d75539e7ea5c9663951062669d101663051ad9f19abd6d2543/slacker-0.13.0.tar.gz
Collecting tinys3
Downloading https://files.pythonhosted.org/packages/59/c4/0641c936d9b1e49d2e4eaaa22601c76c981f6d3c0360bd63660923ed0835/tinys3-0.1.12.tar.gz
Collecting imgurpython
Downloading https://files.pythonhosted.org/packages/d4/76/514defcb05e6d58ded427f57d736ce6dfe879e0188392385a986e8ef3b7b/imgurpython-1.1.7.tar.gz
Collecting Pillow
Downloading https://files.pythonhosted.org/packages/5b/bb/cdc8086db1f15d0664dd22a62c69613cdc00f1dd430b5b19df1bea83f2a3/Pillow-6.2.1.tar.gz (37.7MB)
Collecting humanize
Downloading https://files.pythonhosted.org/packages/8c/e0/e512e4ac6d091fc990bbe13f9e0378f34cf6eecd1c6c268c9e598dcf5bb9/humanize-0.5.1.tar.gz
Collecting minio
Downloading https://files.pythonhosted.org/packages/78/09/fa848f67bb4aeb60b0399fe275cb2049ecc25bbc36477f173df054411f52/minio-5.0.4-py2.py3-none-any.whl (62kB)
Requirement already satisfied: netifaces in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/netifaces-0.10.9-py2.7-linux-armv7l.egg (from Octoslack==1.9.1) (0.10.9)
Requirement already satisfied: sarge in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/sarge-0.1.5.post0-py2.7.egg (from Octoslack==1.9.1) (0.1.5.post0)
Requirement already satisfied: pytz in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/pytz-2019.3-py2.7.egg (from Octoslack==1.9.1) (2019.3)
Collecting pushbullet.py
Downloading https://files.pythonhosted.org/packages/72/91/78912bfb333f412be8f52bba4aa1c9b68d7f612e0fbc9fc5b88b4c894d26/pushbullet.py-0.11.0-py2.py3-none-any.whl
Collecting pathlib2
Downloading https://files.pythonhosted.org/packages/e9/45/9c82d3666af4ef9f221cbb954e1d77ddbb513faf552aea6df5f37f1a4859/pathlib2-2.3.5-py2.py3-none-any.whl
Collecting pushover_complete
Downloading https://files.pythonhosted.org/packages/28/41/b94e91ed986f9d8e10e4591a73f0ce9a4835d1274e0f622a8ce915a0cbf1/pushover_complete-1.1.1-py2.py3-none-any.whl
Collecting rocket-python
Downloading https://files.pythonhosted.org/packages/d2/a5/72256cf53182620aa12e62dd1199098d04854bcdf83e7e165d7f072c72e4/rocket_python-1.2.11-py2.py3-none-any.whl
Collecting matrix_client
Downloading https://files.pythonhosted.org/packages/6b/0b/65dc841fd3d14e7ebc6081bbfce23365a6b2f68cc6ae2ae2d1d7d59570cd/matrix_client-0.3.2-py2.py3-none-any.whl
Requirement already satisfied: pyserial<3.5,>=3.4 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (3.4)
Requirement already satisfied: semantic-version<2.7,>=2.6 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/semantic_version-2.6.0-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (2.6.0)
Requirement already satisfied: pylru<1.3,>=1.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/pylru-1.2.0-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (1.2.0)
Requirement already satisfied: requests<3,>=2.21.0 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/requests-2.22.0-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (2.22.0)
Requirement already satisfied: chainmap<1.1,>=1.0.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/chainmap-1.0.3-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (1.0.3)
Requirement already satisfied: PyYAML<6,>=5.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/PyYAML-5.1.2-py2.7-linux-armv7l.egg (from OctoPrint->Octoslack==1.9.1) (5.1.2)
Requirement already satisfied: websocket-client<0.57,>=0.56 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/websocket_client-0.56.0-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.56.0)
Requirement already satisfied: wrapt<1.12,>=1.11.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/wrapt-1.11.2-py2.7-linux-armv7l.egg (from OctoPrint->Octoslack==1.9.1) (1.11.2)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (1.6.5)
Requirement already satisfied: futures<3.3,>=3.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/futures-3.2.0-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (3.2.0)
Requirement already satisfied: future<0.18,>=0.17.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/future-0.17.1-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.17.1)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/pkginfo-1.5.0.1-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (1.5.0.1)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.7.19)
Requirement already satisfied: tornado==4.5.3 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/tornado-4.5.3-py2.7-linux-armv7l.egg (from OctoPrint->Octoslack==1.9.1) (4.5.3)
Requirement already satisfied: cachelib<0.2,>=0.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/cachelib-0.1-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.1)
Requirement already satisfied: frozendict<1.3,>=1.2 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/frozendict-1.2-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (1.2)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask_Assets-0.12-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.12)
Requirement already satisfied: werkzeug<0.16,>=0.15.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Werkzeug-0.15.6-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.15.6)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/watchdog-0.9.0-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.9.0)
Requirement already satisfied: regex!=2018.11.6 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/regex-2019.8.19-py2.7-linux-armv7l.egg (from OctoPrint->Octoslack==1.9.1) (2019.8.19)
Requirement already satisfied: emoji<0.6,>=0.5.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/emoji-0.5.4-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.5.4)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask_Babel-0.12.2-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.12.2)
Requirement already satisfied: scandir<1.11,>=1.10 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/scandir-1.10.0-py2.7-linux-armv7l.egg (from OctoPrint->Octoslack==1.9.1) (1.10.0)
Requirement already satisfied: monotonic<1.6,>=1.5 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/monotonic-1.5-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (1.5)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (5.2.1)
Requirement already satisfied: markdown<3.1,>=3.0 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Markdown-3.0.1-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (3.0.1)
Requirement already satisfied: psutil<5.7,>=5.6.1 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/psutil-5.6.3-py2.7-linux-armv7l.egg (from OctoPrint->Octoslack==1.9.1) (5.6.3)
Requirement already satisfied: Flask-Principal<0.5,>=0.4 in /home/octoprint/OctoPrint/venv/lib/python2.7/site-packages/Flask_Principal-0.4.0-py2.7.egg (from OctoPrint->Octoslack==1.9.1) (0.4.0)
[... cut ...] tall --record /tmp/pip-record-tETBZq/install-record.txt --single-version-externally-managed --compile --install-headers /home/octoprint/OctoPrint/venv/include/site/python2.7/Pillow Check the logs for full command output.
Running setup.py install for Pillow: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

The system:
OrangePi PC Plus board with Armbian Linux 4.19.62-sunxi (based on Debian Stretch)
OctoPrint 1.3.12

tedder, 2019-11-05:
```
The headers or library files could not be found for zlib,
```
Install `zlib1g`, if it still fails install `zlib1g-dev`.

rafalmag, 2019-11-05:
Thanks @tedder! I did not realize that I need the additional non-python packages. Thanks to your comment I fixed it by installing all missing packages:
```
sudo apt-get install libtiff4-dev libjpeg8-dev zlib1g-dev libfreetype6-dev liblcms2-dev libwebp-dev tcl8.5-dev tk8.5-dev python-tk
```
Reference https://pillow.readthedocs.io/en/latest/installation.html and https://stackoverflow.com/questions/34631806/fail-during-installation-of-pillow-python-module-in-linux .
Issue can be closed.

closed 2019-11-05 by rafalmag