fraschetti/Octoslack #71: OctoPrint 1.4.0rc2 - Slack messages with snapshot do not send

kazibole, 2019-12-04:
Deployed a new image of OctoPi 0.17.0. Restored my OctoPrint 1.3.12 backup file. Upgraded to OctoPrint 1.4.0rc2.

All of my Octoslack settings imported successfully, however any Slack messages with a snapshot fail to send. The octoprint.log shows the following error:

```
2019-12-03 23:13:18,063 - octoprint.plugins.Octoslack - DEBUG - Upload snapshot - snapshot_upload_method: SLACK
2019-12-03 23:13:18,065 - octoprint.plugins.Octoslack - ERROR - Send message error: 'Settings' object has no attribute 'globalGet'
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 2575, in send_slack_message
    hosted_url, error_msgs, slack_rsp = self.upload_snapshot()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 3410, in upload_snapshot
    local_file_path, error_msgs = self.retrieve_snapshot_images()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 3979, in retrieve_snapshot_images
    localCamera = self._settings.globalGet(["webcam", "snapshot"])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 530, in __getattr__
    return getattr(self.settings, item)
AttributeError: 'Settings' object has no attribute 'globalGet'
```

I have also tried with the Imgur snapshot hosting, same error. If I set the snapshot hosting to none, all Slack messages send successfully.

Please let me know if you need any more information or would like additional testing!

ChrisHeerschap, 2019-12-04:
Seeing the same issue with imgur upload. disabling snapshot share fixes the issue.

fraschetti, 2019-12-17:
Hi @kazibole @ChrisHeerschap 

Finally freed up a bit to look into this. I've already submitted a fix - stay tuned and I'll get a release out soon.

kazibole, 2019-12-22:
Working in 1.9.5!

closed 2019-12-22 by kazibole