fraschetti/Octoslack #72: Request Discord Webhook integration

JohnOCFII, 2019-12-09:
Hi Chris,

Our 3D Printing Community on Mattermost is moving to Discord.  I was wondering if you'd be able to integrate a web hook to send OctoSlack output to Discord?  It seems like similar web hook support exists in Discord: https://support.discordapp.com/hc/en-us/articles/228383668-Intro-to-Webhooks

Here is the new Discord: https://discord.gg/QGVbBT

Thanks,

John

fraschetti, 2019-12-13:
Hi John,

I have been slacking on adding discord support; looks like it's finally time.
Do you mind sending me an e-mal (fraschetti.dev@gmail.com) with a new invite? I'll try to catch it before it expires this time. 

Thanks!

fraschetti, 2019-12-19:
Work in progress...

fraschetti, 2019-12-22:
Supported added in 1.9.5 - let me know if you encounter any issues.

closed 2019-12-22 by fraschetti