fraschetti/Octoslack #74: [Feature request] Custom script / event support

irStoopid, 2019-12-23:
It's great having a notification interface on octoprint using slack, but I've been tinkering with the idea of expanding the notification support by triggering custom scripts (eg. python or bash) on a set interval and sending a notification **_Conditionally_** based on the script output.

For example; I'm running octoprint on a spare netbook. Should it get disconnected from mains power and drain the battery it would be great to be able to send notifications based on certain thresholds.
This is something I'd like to be notified of regardless of printing status, and only conditionally send me a notification based on set conditions, in this case battery status.

One could imagine a feature where if the script return code is 0, the script output is sent as text.

Custom system commands are already included in the current feature-set, but only on an interval / progress when printing or on custom Gcode events (when of course printing).
Sending a notification based on a system command output is currently also unconditional.

fraschetti, 2019-12-29:
Hi @irStoopid 

80% of the work required here is determine the best way to get your event to the plugin. Luckily OctoPrint already has a mechanism to support the bulk of this functionality.
http://docs.octoprint.org/en/master/plugins/mixins.html#simpleapiplugin

The above SimpleApiPlugin would allow you to post name/value pairs (in JSON) and generate notifications accordingly. Where things get a bit tricky if where you mentioned wanting to introduce conditional logic (e.g. return code == 0). It seems reasonable to add some conditional logic but we'd want to be careful not to overcomplicate the solution  - if your source is say a shell script, any complicated logic would likely be better served there rather than anything I whip up within the plugin.

here's a rough idea of how this might work:
1. Octoslack (via OctoPrint) supports an API that will accept a JSON payload of name/value pairs
2. Similar to the G-code event handler logic, I'd introduce the ability to define N custom handlers where each handler can apply some simply logic (e.g. run a regex against one name/value pair) and then send a notification with the added support of leveraging additional attributes from the JSON name/value pairs.

Thoughts?

irStoopid, 2019-12-29:
Hi @fraschetti 

Thanks for the quick response on this.
Your proposal seems quite different from what I initially had in mind, but should be at least as suitable for getting an additional notification interface.

I like the idea of having a more generic API which does not require actively calling scripts of one sort or another as this would definately be more secure and less likely to negatively affect performance compared to calling user definable scripts. A downside however could be that it's slightly more work for a user to ensure some process will run in the background and talk to an API provided than it would be compared to providing a script path and and interval at which it should be run.
Your proposal is more versatile however and I'm all for decent generic implementations that can be used by other plugins as well, rather than only small and short running scripts.

You mention moving the conditional logic into the scripts. Having all conditionals outside Octoslack would mean simply not perfoming a json POST and thus not sending a notification when this is not required. That would mean that there would also be no need for user definable handlers, as any non-empty request could be used as is for notifications (might be oversimplified). I.e. as I see it custom handlers need not be used to get this to work. Though ofcourse they would allow for a more easy way of configuration.

All in all, I think that the SimpleApiPlugin method you propose would make a great addition to Octoslack!