fraschetti/Octoslack #75: Octoslack crashes with "out of memory" when second camera is enabled

timur-tabi, 2020-02-01:
I have two cameras attached to my RPi 3B+, a Logitech C270 and a C930e.  I enable the second camera via the instructions here: https://www.makersmashup.com/post/outside-the-box-multiple-webcam-setup-with-octoprint

When I enable the second camera, my print stutters and Octoprint crashes.  The log shows that Octoslack has run out of memory, and I think it takes Octoprint down with it.  Disabling the Octoslack plugin makes the problem go away.
[octoprint.log](https://github.com/fraschetti/Octoslack/files/4143207/octoprint.log)
[octoprint.log](https://github.com/fraschetti/Octoslack/files/4143214/octoprint.log)

fraschetti, 2020-02-01:
Hi @timur-tabi 

Thanks for reporting this issue.
Can you try disabling the Rasberry Pi temperature (uncheck 'Include RasPi temp' in the Octoslack settings) to see if the issue persists?

Thanks!

timur-tabi, 2020-02-02:
Disable the temperature sensor fixes the Python errors in the log, but Octoprint still crashes.

Changing the extra camera URL to logo.png fixes the Octoprint crashes.

When the crash occurs, my RPi runs out of memory (I can't even run 'ls' from the command line) until the oom-killer kicks in and kills processes.  I can then restart Octoprint.  I have a hard time believing that the image is just "too large".  It feels like it's a memory leak.

timur-tabi, 2020-02-02:
BTW, disabling Octoslack altogether eliminates the out-of-memory errors from the Enclosure plugin.

timur-tabi, 2020-03-14:
Since it looks like this issue isn't going to be fixed anytime soon, can you at least give me the ability to specify a different primary webcam, via a URL?  That way, octoslack shows only my second camera.

fraschetti, 2020-04-04:
Hi @timur-tabi 

I've not been able to track down any memory leakes nor have any been reported. Unfortunately Python's internal libraries to call out to the OS to do certain operations (e.g. query the Pi's temperature) require essentially duplicating the entire process's memory temporarily while the command is executing. That's one potential cause (the more memory the core process is using up, the more likely it will be that any plugin, not just Octoslack, may cause an out of memory issue when it issues an OS command); the more plugins active, the more base memory in use. And then of course there's Octoslack's need to do image downloading and processing - which shouldn't be memory intenstive but ultimately depends on the size of the images it's pulling down.

As for your request to disable the primary camera (for Octoslack) and only use the manually provided cameras... that sounds reasonable. I'll look into this. Created: #78

timur-tabi, 2020-04-04:
I have a theory that this might be occurring you provide a streaming url instead of a snapshot url.  I need to run some tests to confirm.

fraschetti, 2020-04-18:
Hi @timur-tabi 

1. Any chance you had time to validate the streaming URL theory?
2. Are you doing any transformations of additional cameras (rotate, flip, etc.)? I did find one potential leak when transformations are in play.

timur-tabi, 2020-04-19:
I wasn't using transformations, but I can't get the second URL to work at all now.  It just won't show up when I send a "status" command.

fraschetti, 2020-04-25:
Hi @timur-tabi 

If there's an error fetching the second camera's image, it'll show up in the octoprint.log but if not, feel free to send along a debug log file for me to examine.

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

timur-tabi, 2020-04-27:
So I'm pretty sure now the problem is when specifying a streaming URL when it should be a snapshot URL.  Octoslack doesn't crash until about 10-20 minutes into a print, so I'm not really keen on debugging this at the moment.  I think it should be pretty easy for you to reproduce.  Can you try it?

fraschetti, 2020-05-24:
Hi @timur-tabi 
The changes for a new release were piling up and I wanted to get Python 3 + New Slack App support out the door asap - which means I didn't end up spending that much time on this issue.

This will stay near the top of my list of items to address when I have time. Thanks for your help diagnosing the issue thus far!

fraschetti, 2021-04-03:
Haven't heard anything on this one in a while. If this still pops up, let me know and we'll take another look.

closed 2021-04-03 by fraschetti