fraschetti/Octoslack #76: coercing to Unicode error

tedder, 2020-03-26:
I'm going to look and patch this, but I want to get the issue here so I can get it out of my paste buffer.
```
ERROR - Slack RTM message processing error: coercing to Unicode: need string or buffer, NoneType found
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 1877, in execute_rtm_loop
    self.process_rtm_message(slackAPIToken, msg)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_Octoslack/__init__.py", line 1981, in process_rtm_message
    elif alternate_bot_id in message_text:
TypeError: coercing to Unicode: need string or buffer, NoneType found
```

tedder, 2020-03-26:
debugging shows these values:
```
bot_id: @U03FTJK82
message_text: (empty str)
alternate_bot_id: None
```

fraschetti, 2020-04-04:
Hi @tedder 

Let me know if you have a patch on the way, otherwise I'll try to tackle this while I'm working on a few other changes.

Thanks!

tedder, 2020-04-04:
Thanks for the nudge. I put it up.

PR #79 (2020-04-04): fix match=None/unicode coerce error, leave better debugging
Fixes #76.

Merry christmas! Easy fix here, once I added the log line to figure out what was null:

    DEBUG - matching B <@U03FTJK82> and MT  and ABI None

I left my little debugging in there. Feel free to bake this change on your end and release with other stuff.
closed 2020-04-04 by fraschetti