fraschetti/Octoslack #77: Not uploading timelapses to Slack

calonmerc, 2020-03-30:
For a while now, all of my Octoprint images have been unable to upload timelapses to Slack. The timelapse rendering completes, I get a notification about it, but there's no URL for it. I've double checked my settings and it is supposed to include the URL.

The big problem is that the local timelapses get removed.

fraschetti, 2020-04-04:
Hi @calonmerc 

Do you mind providing some rough details about your environment?
-What device/OS are you running?
-How are you connecting to Slack (Webhook or API token)?
-Do you mind following the steps below to generate a debug log for analysis?

1. Enable DEBUG logging
2. Reproduce the issue
3. Send Octoprint.log

Instructions to enable DEBUG logging for Octoslack

* Enable DEBUG logging
Ensure the following has been added to /home/pi/.octoprint/logging.yaml  (or appropriate path for your OctoPrint installation)
```
loggers:
  octoprint.plugins.Octoslack:
    level: DEBUG
```

* Restart OctoPrint
```
sudo service octoprint restart
```

For OctoPi releases, the log file will be located in:
/home/pi/.octoprint/logs/octoprint.log

calonmerc, 2020-04-08:
-What device/OS are you running?
--Variety of Pi 3B and 3B+, all running Raspbian from the OctoPi image.
-How are you connecting to Slack (Webhook or API token)?
--Using the API token on all of them. (The same token, if it matters)
-Do you mind following the steps below to generate a debug log for analysis?
-- I will do these after this print finishes.

calonmerc, 2020-04-09:
[octoprint.log](https://github.com/fraschetti/Octoslack/files/4457831/octoprint.log)

chrisfraschetti, 2020-04-09:
Thanks @calonmerc 

It would appear we have ourselves a typo specific to using Slack as the image/video hosting provider. I'm working on a few other changes and I'll fix/test/commit this as time permits.

Question - was this working before (hosting everything via Slack directory) or did you recently switch from something like S3 to Slack?

Thanks again for the debug log.

calonmerc, 2020-04-09:
Probably a year and a half ago I was using imgur or something, but I've been using Slack, with the timelapse uploads, for consistently over a year. I stopped printing much in November, started again at the end of February. That's when I noticed the timelapses weren't showing anymore.

fraschetti, 2020-04-15:
Hi @calonmerc 

I likely broke the Slack asset upload as part of _recent_ release. There were a number of scenarios where the code might try to use an old upload method if the user was swapping around configurations. The logic to ensure the setup was correct for Slack uploads had a typo.
I've submitted a fix which will be included in my next release.

calonmerc, 2020-04-15:
Awesome! Any ETA on the release? Or any help running dev code? (I don't see a development branch to pull)

fraschetti, 2020-04-21:
Hi @calonmerc 
Shouldn't be long before I push a new release - likely within the next week or so.
As far as the dev branch is concerned, this project has been fairly simple thus far and I haven't needed to break out a dev branch from the main branch.  If you want to test the change, you can simply pull the main branch and go from there. If you haven't seen this page before, the steps to register a development plugin can be found here - https://docs.octoprint.org/en/master/plugins/gettingstarted.html

brad-colbert, 2020-05-06:
May I "pile on" this thread?

I'm not seeing either snapshots or timelapses posted to Slack.  I'm assuming that Octoslack is able to post both directly?  If not I apologize for not understanding.

Cheers!

fraschetti, 2020-05-24:
Hi @calonmerc 
I've pushed new Octoslack 2.0.0 release which should address your Slack upload issue. Give it a try - if you still experience issues, feel free to reopen this ticket.

fraschetti, 2020-05-24:
Hi @brad-colbert 

Assuming you're config is also setup to use Slack to host the snapshots/timelapses, the new release should fix your issue as well. It you were using a different hosting method or are experiencing a different issue, feel free to open another ticket that we can use to investigate your issue further.

closed 2020-05-24 by fraschetti

brad-colbert, 2020-05-24:
Yep, testing now.  Will know in roughly 3.5 hours!   Thank you for all of
your hard work!  Fantastic plugin!

Kind regards,
Brad