fraschetti/Octoslack #78: Allow snapshot images to ignore the primary Octoslack camera

fraschetti, 2020-04-04:
Allow snapshot messages to only include images from the manually provided camera URLs and not the default camera configured in Octoslack

timur-tabi, 2020-04-04:
What I would prefer is to simply be able to specify and number of camera URLs as 1, 2, 3, etc.  By default, Octoslack would configure the first camera as normal, but that configuration would be editable.  And then add an enable/disable checkbox next to each entry.

fraschetti, 2020-04-18:
Hi @timur-tabi 

The use case being you'd want to use different cameras for different events?

timur-tabi, 2020-04-18:
Different cameras for different events would be neat, but I don't need that feature.  I'm just asking for the ability to specify the primary camera by URL.  I have two cameras attached to my Pi, but the Octoslack config page does not give me an option to specify a camera.  And for some reason, Octoslack is now using my second camera.  It used to use my first camera, but somehow it magically switched to my second.