fraschetti/Octoslack #81: [Feature request] Octolapse support

N-Schaef, 2020-04-16:
Hello,

I've noticed that octolapse timelapses are not triggering a notification and/or upload of the timelapse.

After further investigation I saw that octolapse decided against using the native `MovieDone` trigger.
Instead they use a plugin-specific `PLUGIN_OCTOLAPSE_MOVIE_DONE` trigger.
Appart from that the payload seems to be the same. 
At least it contains the required `movie` attribute with the timelapse path.

So my guess is that to support this, all 
``` js
event == "MovieDone"
```
would have to be replaced by

``` js
event == "MovieDone" || event == "PLUGIN_OCTOLAPSE_MOVIE_DONE"
```

I can also test it when I will do my next print and do a pull request.

fraschetti, 2020-04-18:
Hi @N-Schaef 

Sound like a good change. One suggestion - you might want to add your logic to on_event or handle_event and instead of adding an or clause, simply look for PLUGIN_OCTOLAPSE_MOVIE_DONE and update event to the expected MovieDone. This will save myself and others a bit of headache down the road as we won't have to deal with a special case. I suppose what matters most is whether or not Octolapse's event payload is the same or if it differs and we need to do a transofmration or all special logic to handle its unique payload.

Good luck - let me know if you run into any issues,

N-Schaef, 2020-04-20:
@fraschetti good idea.

I implemented it (#82) and tested the change with Octolapse v0.4.0rc1.
I think the current stable version does not yet have the event.

fraschetti, 2020-04-21:
Thanks @N-Schaef 

I've merged the change.

closed 2020-04-21 by fraschetti