fraschetti/Octoslack #83: Event - build plate temperature

Matt2k34, 2020-04-24:
Hey
Set this up yesterday and loving it so far. 

I'm not sure if this is supported and I'm missing something, or isn't supported. 

Can you add events for hotend / hotbed cooling? It's great that it prompts on complete, I just have to give it 10-15 mins for the glass bed to cool so the print pops off, would be good to either add a followup notification on a timer. Or better if it can pull the hotbed temp and report on that. (eg when the hotbed passes below X degrees, - and possibly with a margin of error for PID fluctuations) 

Thanks!

Wob76, 2020-05-28:
This would be very nice, I saw this plugin come up today, it triggers GCODE, could use it add a monitor in octoslack for a specific M300 Gcode, but direct support would be nice.

https://github.com/gmccauley/OctoPrint-CooldownNotification

Wob76, 2020-05-28:
I just did a quick test, I installed the CooldownNotification and set the Execute GCODE to "M300 S500 P500" (a simple beep) and then configured Octoslack to Monitor for G-code sent starts with "M300 S500 P500" and it works as expected and includes the print temperatures too.

It is 2 plugins to get it done, but it is a solution for now.

M1S2, 2020-05-28:
Hi, your plugin is working like a charm. But it would be really cool to have events on reaching some temperature thresholds on all hotends. E.g. for heat up done or cool down done.

fraschetti, 2020-05-29:
Hi @Matt2k34 @Wob76 @M1S2 

This shouldn't be all that difficult to implement. I have an idea or two how to enhance what I saw in CooldownNotification 

@M1S2 - I've only ever used a single nozzle printer and therefore have only seen internal OctoPrint data for a single nozzle. You mentioned "nozzles" - do you have any insight into multi-nozzle printing? I'm trying to formulate how to best handle multiple nozzle with different temperature expectations (different target temps for different materials, not alll nozzles are active, etc.)

M1S2, 2020-05-29:
Hi, sorry, I only have a single nozzle on my printer. With all hotends I mean the nozzle and heatbed.
So it would be cool to receive a notification if the heatbed and nozzle reach their target temperatures (heating done) or if the heatbed falls below a specific threshold (ready to remove the print from the heatbed).