fraschetti/Octoslack #84: Octoslack Not Listening

Llama-Trauma, 2020-04-30:
After configuring the App, Slack is successfully sending automated print messages; however, none of the commands sent back through slack are triggering a response. 

I have "Listen for commands" and "Use Slack identity" checked. 'Help' and 'Status' do not respond. 
[image]

My Slack names are all over the place, but the only 'Active' App I see in Slack for mentions is MK2.
[image]

Also curious why I only see "Print started: {print_name}, Estimate: {remaining_time}" under Fallback? If I want to remove the 'User' from the print started dialogue, is there a way to customize this further?
[image]

Thank you for your support!

Humsie, 2020-05-02:
i struggled with the same problem last week, and also found a lot of RTM login failures in the log file.
Turns out octoslack uses the "old" rtm api which requires a "classic" slack bot. 

i created a classic bot using the link from: https://api.slack.com/rtm#classic
and used the "Bot User OAuth Access Token" in octoslack and status command worked. (and the failures stopped in the log file :))

fraschetti, 2020-05-02:
Hi @Llama-Trauma and @Humsie 

@Humsie is indeed correct - Octoslack currently leverages the older/classic bot support within Slack which supports RTM (websockets). Unfortunately the new Slack bots/apps are a bit different (for good reason) but unfortunately RTM is no longer supported for new bots.
For now, I'd rely on the workaround @Humsie listed above but I am currently working on adding proper support for the bot/app connectivity requirements. 

Stay tuned.

fraschetti, 2020-05-24:
Hi @Llama-Trauma and @Humsie 

I've just released Octoslack 2.0.0 which includes proper support for the new Slack Apps/Bots.  This includes new logic to move away from the RTM functionality only available to classic Slack Apps/Bots.

What scopes/permissions you need will depend on what you want to do with the plugin. I've included the list of scopes within the plugin settings dialog but I'll also share them here for initial reference. This list may change over time as Slack makes changes or as we add additional functionality to the plugin. The Octoslack settings dialog should be see as the source of truth going forward.

### **Slack Bot Webhook Scopes**
Bot Token Scopes requied for posting to a channel
- channels:read
- chat:write
- groups:read

Bot Token Scopes requied for uploading assets/files to Slack
- files:write

### **Slack Bot Access Token Scopes**

Bot Token Scopes requied for posting to a channel
- channels:read
- chat:write
- groups:read

Bot Token Scopes requied for uploading assets/files to Slack
- files:write

Bot Token Scopes requied for listening for and responding to commands
- channels:history
- groups:history
- reactions:write
- users:read

closed 2020-05-24 by fraschetti

Llama-Trauma, 2020-06-03:
Thanks for the update, Chris! 

Would you be able to direct me towards a guide for setting up the Slackbot side of this plugin? I am able to receive automated print notifications still, but I cannot get any of the live Slackbot commands to work for checking status, or affecting the print.