fraschetti/Octoslack #85: [feature request] thunbnail from gcode

JFSolution, 2020-05-04:
Include thumbnail from gcode in print started message

Here is how to include a thumbnail in gcode:
https://plugins.octoprint.org/plugins/prusaslicerthumbnails/

jneilliii, 2020-08-02:
@fraschetti Both the PrusaSlicerThumbnails plugin and the UltimakerFormatPackage plugin utilize the same approach for thumbnails. They both add `thumbnail` and `thumbnail_src` metadata properties to the files, which is returned by the file API, and also accessible via [self._file_manager.get_metadata("local", <file_path>)](https://docs.octoprint.org/en/1.4.1rc4/modules/filemanager.html#octoprint.filemanager.storage.LocalFileStorage.get_additional_metadata) calls in Python if I'm not mistaken.

charles4308, 2020-08-02:
@fraschetti I think if you can store a second variable with the file name, without the '.gcode' at the end, this would work as a quick fix.

Ultimaker Preview Url:
http://<IP_ADDRESS>/plugin/UltimakerFormatPackage/thumbnail/<FILENAME>.png

I think your {print_name} variable contains the file extenstion.

If this were changed, i would could replace <FILENAME> in the url with your stored variable.

here is an example when i manually enter the URL:
[image]