fraschetti/Octoslack #86: Appears that I'm now longer able to get token for Slack

brad-colbert, 2020-05-05:
Hello,

I think I followed the instructions correctly for a Slack integration but there is now this message on Slack:

"We recommend replacing your custom integrations with Slack apps, which have more features and use the latest APIs."

There appears to be no other option.
[image]

Is there another way?

Thank you,
Brad

brad-colbert, 2020-05-05:
Per the instructions:

"Apps & integrations --> Manage --> Custom Integrations --> Bots"

This no longer is an option.

tedder, 2020-05-05:
Brad, try this:
* create an app: https://api.slack.com/apps?new_app=1
* under "add features", click "bots"
* "review scopes"
* under "bot token scopes", add "incoming-webhook", "channels:join", "chat:write" and "chat:write.public". (probably need more if you are going to reply to it, I think this should be good for notifications)
* now scroll to the top, "install app to workspace"

This is just a guess, I'm going to try it now and see how it works, but I only use "notifications", not commands, so more testers are needed.

brad-colbert, 2020-05-06:
Ted,

That worked!  Thank you!

tedder, 2020-05-06:
brad or @fraschetti can you reopen this so we update the documentation or figure out an easier way to do this?

fraschetti, 2020-05-06:
Hi @tedder 
I'll reopen it so you and others can easily find this conversations and comment as necessary.

Good news - I've already finished the development of support for the new supported methods for listening for slack commands (for this plugin, that will be polling Slack's endpoint). I'm integrating those changes right now. Stay tuned.

brad-colbert, 2020-05-06:
Ok, sorry.

brad-colbert, 2020-05-06:
Sorry, I got your message later in the morning.  Can't wait for the updates!  Thank you for the hard work!

davidkmiller, 2020-05-12:
Can anyone advise what is required under "bot token scopes" in Slack to allow commands; I've used this thread to get notifications working, but I'm not sure what to do for commands? Thanks

fraschetti, 2020-05-24:
Hi @brad-colbert @tedder and @davidkmiller 

I've just released Octoslack 2.0.0 and with that comes proper support for the new Slack apps. What scopes/permissions you need will depend on what you want to do with the plugin. I've included the list of scopes within the plugin settings dialog but I'll also share them here for initial reference. This list may change over time as Slack makes changes or as we add additional functionality to the plugin. The Octoslack settings dialog should be see as the source of truth going forward.

### **Slack Bot Webhook Scopes**
Bot Token Scopes requied for posting to a channel
- channels:read
- chat:write
- groups:read

Bot Token Scopes requied for uploading assets/files to Slack
- files:write

### **Slack Bot Access Token Scopes**

Bot Token Scopes requied for posting to a channel
- channels:read
- chat:write
- groups:read

Bot Token Scopes requied for uploading assets/files to Slack
- files:write

Bot Token Scopes requied for listening for and responding to commands
- channels:history
- groups:history
- reactions:write
- users:read

closed 2020-05-24 by fraschetti